autowarefoundation/autoware #747: Yolo crashing during runtime

robotvin, 2017-07-26:
Hi
I have recently installed the yolo node.. and installed successfully..
During runtime it is crashing with Code 11 error... (memory leak) 

When i debugged the code... I found it crashing at following line.
line no 100 in yolo2.cpp - float *prediction = network_predict(darknet_network_, in_data); 

and one more i have observed the following function has been called with 12 arguments but it requires 13. 

line 108 in yolo2.cpp - get_region_boxes(output_layer, in_darknet_image.w, in_darknet_image.h,
							darknet_network_.w, darknet_network_.h,
							min_confidence_, darknet_box_scores_.data(), darknet_boxes_.data(),
0, 0, 0.5, 1); 

Kindly go through the code again.. suggest corrections...

Thanking you

amc-nu, 2017-07-26:
Regarding that error 
I believe you're using a different version of Yolo2
Make sure to follow the README from the node.
Recently the Yolo author updated the API

robotvin, 2017-08-03:
Thanks for comments
I have installed the new version of yolo2. but still it is crashing. its not able to initialize the Darknet. Please can please suggest as i am not able to find the bugg.

robotvin, 2017-08-03:
I have attached screen shot of terminal. plz guide # @amc-nu 
[image]

dejanpan, 2017-08-03:
@robotvin Can you also comment: 
- which ubuntu version, 
- which ROS version, 
- which autoware version, 
- how can we reproduce your error.

thx

robotvin, 2017-08-03:
which ubuntu version - 14.04
which ROS version - ROS INDIGO
which autoware version - downloaded 10 days back
how can we reproduce your error - you can run bag file with video data and run yolo2 detector

dejanpan, 2017-08-03:
Please provide exact git sha commit that you downloaded 10 days ago (git log command).

Please also provide short version of a bag file with which one you test and **exact** commands that you run yolo2 detector with.

robotvin, 2017-08-03:
Thank you sir
I think its not the issue with the way i am running or launching yolo. But its clearly indicating memory access problem/allocation issue in yolo2.cpp. Its failing when in load function for darknet or in covert_image function.

dejanpan, 2017-08-03:
We are trying to reproduce and debug your problem. And we can't do that if
you don't provide exact steps for when the error happens.

yukkysaito, 2017-08-06:
@robotvin What is the GPU you are using? I think that you need at least 4 GB of GPU memory size

closed 2017-08-29 by amc-nu

robotvin, 2017-09-01:
sorry for late reply..as i was on vacation and away from my work....
I have GPU with 8GB memory. The issue is the yolo is working fine on following resolutions
640x480, 768x576 and 1280x960. But for 658x492 (which is my data) is crashing. Is there some guidelines for network-dimentions and video resolution. plz help...