autowarefoundation/autoware #755: Release v.1.4.0

yk-fujii, 2017-08-01:
We will release Autoware v.1.4.0.
----
## Target date: 4  Aug. 2017
Even if not all PRs are in time, we will release on this date, since the merger of #753 to the master takes precedence.
## Target issues and PRs
- [x] #741 
- [x] #742
- [x] #746 
- [x] #748 
- [x] #753 
- [ ] #733 
- [ ] #749 
- [ ] #751 
----

### Next releasing targets (v.1.5.0)
- [ ] #727 
- [ ] #739 
- [ ] #754

dejanpan, 2017-08-01:
@yk-fujii what do you think of a release wiki page?

It could be as simple as this one: https://github.com/ros2/ros2/wiki/Releases.

For every release we can have three categories:
- New Features
- Demos/Documentation
- Known Issues

yk-fujii, 2017-08-02:
Is there any benefit to having it prepared separately from the release page?
If there is, I think that adoption is good.
https://github.com/CPFL/Autoware/releases

Also, categorizing is good for easily understand release.

dejanpan, 2017-08-02:
Ahh never mind, didn't notice: https://github.com/CPFL/Autoware/releases. great job.

Maybe yeah, just to add those 3 categories.

dejanpan, 2017-08-04:
@yk-fujii could you maybe also include this PR: https://github.com/CPFL/Autoware/pull/760 in this release and then use this command to make a new release tag: http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease#bloom.2BAC8-Tutorials.2BAC8-PrepareUpstream.Update_package.xml_Version?

yk-fujii, 2017-08-04:
Done.
Thanks for helping us!

closed 2017-08-04 by yk-fujii