autowarefoundation/autoware #758: Test Gazebo 7.0 and integrate Gazebo 8.0

dejanpan, 2017-08-03:


dejanpan, 2017-08-03:
a) try out and document current state of gazebo integration in Autoware
b) port gazebo 8.0 to Autoware
c) prepare a demo in Gazebo 8.  We could do the same as here (https://github.com/osrf/car_demo) but have the car follow waypoints on that track? 
d) document

yukkysaito, 2017-08-06:
Great!

huckl3b3rry87, 2017-11-02:
This sounds great, I am interested in testing my path planning code and this seems like Autoware will be a nice testbed for my algorithm. Currently I am modeling the vehicle, and sensors in Gazebo and would like to use some Gazebo environment to collect the data for my simulations (instead of rosbag). @dejanpan are you still planning on working on this?

Thanks!

dejanpan, 2017-11-10:
@huckl3b3rry87 unfortunately there was no time to work on this. We would however warmly welcome everyone willing to chip in. Do you have some spare time?

huckl3b3rry87, 2017-11-10:
@dejanpan I do have some time to work on this; I decided (at least for the moment) that I will use Autoware for my Autonomous Vehicle Proving Grounds.

dejanpan, 2017-11-11:
@huckl3b3rry87 great. I would then suggest the following:

1. port OSRF's docker into Autoware docker:  https://github.com/CPFL/Autoware/tree/master/docker
1. Make sure we can drive the Prius using joystick: https://github.com/CPFL/Autoware/blob/fad2d17234e77e393e263aac861b7a835f10903a/ros/src/system/gazebo/catvehicle/launch/joystick.launch
1. Create an NDT map: 
[Autoware_NDT_Mapping_Tututorial.pdf](https://github.com/CPFL/Autoware/files/1463215/Autoware_NDT_Mapping_Tututorial.pdf)

This would be a great base for all the further work.

Let me know if you need assistance/help with any of the steps.

huckl3b3rry87, 2017-11-11:
@dejanpan for #2, you would like to integrate the prius vehicle from car_demo into Autoware?

dejanpan, 2017-11-11:
@huckl3b3rry87 - yes, that is correct

dejanpan, 2017-11-27:
@huckl3b3rry87 have you been able to make some progress here? Got any questions?

huckl3b3rry87, 2017-11-27:
@dejanpan I did put some time into it a couple weeks ago, but had to shift priorities a bit. This is still definitely something that I am interested in and will put some time in today!

huckl3b3rry87, 2017-11-28:
@dejanpan today, I did my best to update the Docker file as well as the build and run scripts. The result is here https://github.com/huckl3b3rry87/Autoware/blob/feature/port_car_demo/docker/Dockerfile

After building a docker image with:
```
$/Autoware/docker/ ./build_demo.bash
```
and trying to run the demo with:
```
$/Autoware/docker/ ./run_demo.bash
```
The gui pops up for about a second and disappears. I have spent quite a bit of time on this, but cannot figure it out. Any help would be greatly appreciated! Thanks!

huckl3b3rry87, 2018-02-21:
@dejanpan I have continued to work on this project and actually got Gazebo 8 working with the latest Autoware master, using [this docker file](https://github.com/huckl3b3rry87/Autoware/blob/feature/port_car_demo/docker/Dockerfile.test) by working I mean it pops up Gazebo from Autoware and I can see the cat_vehicle. The problem is that Gazebo is running very slowly with a real time factor of about 0.15, and Gazebo is slow even without Autoware. It seems that someone else had an issue [here](https://groups.google.com/forum/#!topic/autoware/hjmb5moP8qs) where there is a [pull request](https://groups.google.com/forum/#!topic/autoware/hjmb5moP8qs) for Gazebo that was mentioned for the fix. 

Also, the in this branch is a [docker test file](https://github.com/huckl3b3rry87/Autoware/blob/feature/port_car_demo/docker/Dockerfile#L4) that I can use to install an image of car_demo that runs with a real-time factor of 1. @tfoote I basically coppied all of the Docker files that where upstream from yours so that we could build ``FROM nvidia/cuda:8.0-devel-ubuntu16.04``. Also every line has the code where I got it from and comments with H.F. if I changed something. 

So, somehow installing additional packages for Autoware is realling slowing down gzserver. Any ideas or suggestions @tfoote would be very helpful. Thanks!

huckl3b3rry87, 2018-02-21:
**Update**
Using (feature/port_car_demo)[https://github.com/huckl3b3rry87/Autoware/tree/feature/port_car_demo] branch I am able to build on my machine (Docker image still needs a little though, I think on how to actually build it) Autoware and car_demo (launched when the Gazebo button is clicked). By building it with this command ``catkin_make --pkg prius_msgs && catkin_make``.

huckl3b3rry87, 2018-02-22:
**Update**
I am now able to run the car_demo package using Docker within Autoware. Also, I am working on building it on Docker hub https://hub.docker.com/r/huckl3b3rry87/autoware/builds/

yukkysaito, 2019-02-16:
https://github.com/CPFL/Autoware/pull/1930
This issue will be closed to replace the new Gazebo
Ok?

closed 2019-02-22 by yukkysaito