autowarefoundation/autoware #763: Release v.1.5.0

yk-fujii, 2017-08-04:
We will release Autoware v.1.5.0.
----
## Target date:  31 Aug. 2017
## Target issues and PRs
- [ ] #727 
- [ ] #739 
- [ ] #754 

If you have any commits and you want to put in, please tell me here.
----
### Next releasing targets (v.1.6.0)

dejanpan, 2017-08-26:
@yk-fujii can you run the following commands when you will be releasing such that we get consistent changelogs and version numbers
```
catkin_generate_changelog
catkin_prepare_release
```

This is from here: http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease#bloom.2BAC8-Tutorials.2BAC8-PrepareUpstream.Update_Changelogs.

yk-fujii, 2017-09-03:
I will postpone this work a bit. . .

yk-fujii, 2017-09-20:
We will release it on September 21.

closed 2017-09-21 by yk-fujii