autowarefoundation/autoware #765: Why does Autoware need root privileges?

therewillbecake, 2017-08-05:
Hello,

I'm currently trying to find out why Autoware requests the root password when starting the runtime manager since I might not have root privileges on the computer on which I want to run Autoware.

I think I narrowed it down to the "proc_manager.py" script in /Autoware/ros/src/util/packages/runtime_manager/scripts/. It seems to open a socket to communicate with the runtime manager. When I run proc_manager.py without sudo I get an error
`[Errno 98] Address already in use`
and the socket cannot be opened. However, the rest of Autoware seems to run just fine, despite proc_manager.py not working.

So here are my questions:
- Is proc_manager.py the only part of Autoware that needs root privileges?
- Does the proc_manager need them only to open the socket or anything else?
- What does the proc_manager do and what parts of Autoware depend on it?

Thank you for your time and consideration.

dejanpan, 2017-08-05:
@yk-fujii @kuriking @syohex maybe one of you could answer this? I was also wondering but never had time to look into the code.

yukkysaito, 2017-08-06:
@therewillbecake @dejanpan Because it is necessary for real-time processing. For example, to change the priority of the process (ROS node) and scheduling algorithms. If you do not process in real time, it is not essential.

dejanpan, 2017-08-06:
So are you changing `nice` values on the PID of ros nodes? Is there an example of that?
Is there any other reason?

yk-fujii, 2017-08-06:
Autoware change the nice value and also change the process scheduling policy.
https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/proc_manager.py#L102-L119
In addition, root may be necessary for the execution of the USB-UVC driver.

We should fix as Autoware ask for root privilege on each occasion (such as when the sys button is pressed).

dejanpan, 2017-08-08:
this explains it for me. I added the answer to FAQ (https://github.com/CPFL/Autoware/wiki/FAQ(EN)#why-does-autoware-need-root-privileges) and will close the issue if more explanation is needed.

closed 2017-08-08 by dejanpan