autowarefoundation/autoware #766: Release further pkgs into ROS buildfarm

dejanpan, 2017-08-05:
Couple of more things to consider:
- pkg names (there are other pkgs in ROS with the same name, e.g. velodyne_msgs)
- update maintainers in package.xml, also add Dejan everywhere
- find which pkgs to release
- group all pkgs under autoware namespace, see: http://wiki.ros.org/geometry?distro=lunar

dejanpan, 2017-08-06:
Had to also first fix autoware_msgs: http://build.ros.org/job/Kbin_uX64__autoware_msgs__ubuntu_xenial_amd64__binary/1/
See example cmake and package.xml file: https://github.com/ros/common_msgs/tree/jade-devel/stereo_msgs

dejanpan, 2017-08-06:
For ${catkin_EXPORTED_TARGETS}: http://wiki.ros.org/catkin/CMakeLists.txt#Important_Prerequisites.2BAC8-Constraints

dejanpan, 2017-08-08:
For pre-release jobs ():

* drop CATKIN_IGNORE files into all the other packages on your desired test branch

* pre-release website: http://prerelease.ros.org/kinetic

*
``` 
generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   kinetic default ubuntu xenial amd64 --custom-repo     Autoware:git:https://github.com/CPFL/Autoware:feature/buildfarm_test_branch   --level 1   --output-dir ./
```

dejanpan, 2017-08-26:
Complete release instructions:

1. Fix CMakeLists.txt and package.xml
Common issues (see https://github.com/CPFL/Autoware/pull/770/files for example):
- Install targets are not specified
- dependencies in find_package and catkin_package are missing (in CMakeLists.txt)
- dependencies are missing for build_depend and run_depend in package.xml
- some dependencies are superfluous, that is, you need to check in the source code if they are really needed
- ${catkin_EXPORTED_TARGETS}: http://wiki.ros.org/catkin/CMakeLists.txt#Important_Prerequisites.2BAC8-Constraints

2. Create a pre-release
- Install build farm: http://wiki.ros.org/regression_tests#How_do_I_setup_my_system_to_run_a_prerelease.3F

* drop CATKIN_IGNORE files into all the packages that you do NOT want to release, push to buildfarm_test_branch (see why https://github.com/ros/rosdistro/pull/15705#issuecomment-320744956) 
 -    mkdir -p /tmp/prerelease_job
- cd /tmp/prerelease_job
- generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   kinetic default ubuntu xenial amd64 --custom-repo     Autoware:git:https://github.com/CPFL/Autoware:feature/buildfarm_test_branch   --level 1   --output-dir ./

3. Make release
Once everything compiles in step 2 you can run the following commands:
catkin_generate_changelog
catkin_prepare_release
bloom-release --rosdistro <ros_distro> --track <ros_distro> repository_name --edit

amc-nu, 2017-08-28:
@dejanpan I'll take care of 
* pointgrey
* points_downsampler
* lidar_detector
* velodyne_pointcloud

@kitsukawa can you please fix 
* ndt_localizer

@TomohitoAndo could you check:
* astar_planner
* waypoint_maker
* lane_planner

@hatem-darweesh would you verify:
* way_planner
* dp_planner

@kuriking and @kondoh2 can you help with these ones?
runtime_manager
model_publisher
autoware_connector
road_wizard

@shinpei0208 could you suggest who can check these nodes? (if Axe is not available)

Thanks all for your help

dejanpan, 2017-09-02:
I did `model_publisher`, PR will follow.

dejanpan, 2017-09-02:
I did `autoware_connector`, PR will follow.

dejanpan, 2017-09-02:
did libvectormap

sumanth-nirmal, 2018-04-16:
The packages mentioned above and their dependencies are fixed for release. The CMakeLists.txt and package.xml are cleaned and fixed for missing dependencies, added install targets etc. The complete list of packages fixed are as below:
 - amathutils
 - autoware_msgs
- autoware_connector
 - autoware_pointgrey_drivers
 - custom_msgs
  - dbw_mkz_msgs
  - filters
  - gnss
  - image_processor
  - model_publisher
  - as
 - ndt_cpu
  - ndt_gpu
  - ndt_tku
  - registration
  - state
  - tablet_socket_msgs
  - runtime_manager
  - tablet_socket
  - vector_map_msgs
  - vector_map
  - decision_maker
  - libvectormap
  - map_file
  - vector_map_server
  - velodyne_msgs
  - velodyne_driver
  - velodyne_pointcloud
  - ndt_localizer
  - points_downsampler
  - points_preprocessor
  - waypoint_follower
  - astar_planner
  - lane_planner
  - road_wizard
  - waypoint_maker

All these are checked as per https://github.com/CPFL/Autoware/issues/766#issuecomment-325096495 and tested as per step 2 in https://github.com/CPFL/Autoware/issues/766#issuecomment-325096495. (check this for any remaining TODO's https://github.com/sumanth-nirmal/Autoware/wiki/Packages-fix-update) Remaining packages have `CATKIN_IGNORE`

closed 2020-05-29 by gbiggs

Timple, 2020-07-09:
Why is this closed?

Packages are fixed (as per last [comment](https://github.com/Autoware-AI/autoware.ai/issues/766#issuecomment-381532160)) but not released as per title.
(I cannot install ros-melodic-points-preprocessor)