autowarefoundation/autoware #771: Duplicate pkg names in Autoware

dejanpan, 2017-08-08:
In Autoware there are couple of packages that have the same or similar name as released packages in ROS. I listed them below as pointer to ROS (original driver), Autoware driver and proposal how we should rename them. 
Released package names in ROS must be unique. Can you please review and if you agree also go ahead and rename?

ROS: https://github.com/ros/filters
Autoware: https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/localization/lib/fast_pcl/filters
Proposal: https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/localization/lib/fast_pcl/pcl_filters

ROS: https://github.com/ros-drivers/microstrain_3dmgx2_imu
Autoware: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/imu/packages/microstrain
Proposal: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/imu/packages/autoware_microstrain_3dm_gx5_45

ROS: https://github.com/ros-drivers/nmea_navsat_driver
Autoware: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/gnss/packages/nmea_navsat
Proposal: 
a) https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/gnss/packages/autoware_nmea_navsat
b) Move into some more generic upstart pkg since it only contains one launch file.

ROS: https://github.com/ros-drivers/pointgrey_camera_driver/tree/master/pointgrey_camera_driver
Autoware: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/camera/packages/pointgrey
Proposal: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/camera/packages/pointgrey_camera_driver_grasshooper3_ladybug

ROS: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/imu/packages/xsens/src/xsens_driver
Autoware: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/imu/packages/xsens/src/xsens_driver
Proposal: https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/imu/packages/xsens/src/autoware_xsens_driver

weisongwen, 2017-11-02:
@dejanpan @shinpei0208

weisongwen, 2017-11-02:
Hi, @dejanpan @shinpei0208 , sorry to bother you here.
# Bug
## Expected Behavior
compile Autoware package successfuly

## Actual Behavior
compile error as following:
**CMakeFiles/icp_matching.dir/nodes/icp_matching/icp_matching.cpp.o: In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::align(pcl::PointCloud<pcl::PointXYZ>&, Eigen::Matrix<float, 4, 4, 0, 4, 4> const&)':
/usr/local/include/pcl-1.8/pcl/registration/impl/registration.hpp:205: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::setPointRepresentation(boost::shared_ptr<pcl::PointRepresentation<pcl::PointXYZ> const> const&)'
CMakeFiles/icp_matching.dir/nodes/icp_matching/icp_matching.cpp.o: In function `pcl::Registration<pcl::PointXYZ, pcl::PointXYZ, float>::Registration()':
/usr/local/include/pcl-1.8/pcl/registration/registration.h:127: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
/usr/local/include/pcl-1.8/pcl/registration/registration.h:127: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
CMakeFiles/icp_matching.dir/nodes/icp_matching/icp_matching.cpp.o: In function `CorrespondenceEstimationBase':
/usr/local/include/pcl-1.8/pcl/registration/correspondence_estimation.h:104: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
/usr/local/include/pcl-1.8/pcl/registration/correspondence_estimation.h:104: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::KdTree(bool)'
CMakeFiles/icp_matching.dir/nodes/icp_matching/icp_matching.cpp.o: In function `pcl::registration::CorrespondenceEstimationBase<pcl::PointXYZ, pcl::PointXYZ, float>::initComputeReciprocal()':
/usr/local/include/pcl-1.8/pcl/registration/impl/correspondence_estimation.hpp:112: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::setPointRepresentation(boost::shared_ptr<pcl::PointRepresentation<pcl::PointXYZ> const> const&)'
CMakeFiles/icp_matching.dir/nodes/icp_matching/icp_matching.cpp.o: In function `pcl::registration::CorrespondenceEstimationBase<pcl::PointXYZ, pcl::PointXYZ, float>::setInputTarget(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&)':
/usr/local/include/pcl-1.8/pcl/registration/impl/correspondence_estimation.hpp:74: undefined reference to `pcl::search::KdTree<pcl::PointXYZ, pcl::KdTreeFLANN<pcl::PointXYZ, flann::L2_Simple<float> > >::setPointRepresentation(boost::shared_ptr<pcl::PointRepresentation<pcl::PointXYZ> const> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wenws/Autoware/ros/devel/lib/icp_localizer/icp_matching] Error 1**

## Steps to Reproduce the Problem

  1.download the Autoware of master branch
  2.compile with ./catkin_make_release
  3. Error arise when compile to 98%

## Specifications

  - Ubuntu version:14.04 LTS
  - ROS Version: Jade
  - Autoware branch: newest master branch
  -PCL version: 1.8

# New Feature
Actually, I upgrade my PCL version from PCL 1.7 to PCL 1.8 for some reason, I think this can be the possible reason, but how can Autoware be compatible with PCL 1.8?
[image]

dejanpan, 2017-11-09:
@weisongwen can you please create a new issue? You have now commented in one of my issues that has nothing to do with your error. Then I will answer there.

weisongwen, 2017-11-13:
Hi, @dejanpan  sorry for the inconvenience that brought to you here, and thanks a lot for your kind reply.
I do not know why I can not create issue in the "Autoware issue"....
 I will try to create another issue later.

closed 2020-05-29 by gbiggs