autowarefoundation/autoware #773: Qt5Core linking Issue

dejanpan, 2017-08-08:
There are several packages in Autoware that require Qt5Core library to link even though they do not use use Qt at all, e.g. https://github.com/CPFL/Autoware/blob/master/ros/src/sensing/filters/packages/points_preprocessor/CMakeLists.txt#L18.

The problem seems to be with the bug in PCL which brings in Qt: https://github.com/ros-perception/perception_pcl/pull/151.

I suggest to do the same fix. @amc-nu let me know if you need help.

dejanpan, 2017-08-09:
Mikael, 
I didn't know where to put this question so I am just shooting you an email but will also put it in the open if you wonna answer there: https://github.com/CPFL/Autoware/issues/773.

I am releasing Autoware pkgs as you know and I had the issue that a package failed on linking by not finding Qt5Core, e.g. https://github.com/CPFL/Autoware/blob/develop/ros/src/data/packages/map_file/CMakeLists.txt.

Then  we came across the issue that you solved here: https://github.com/ros-perception/perception_pcl/pull/151 which we applied and that fixed it. I like the real solution by not linking against the library that you do not use.

However now I see that we have two situations:

1) Package finds, compiles and links against PCL library only e.g.:  https://github.com/CPFL/Autoware/blob/develop/ros/src/data/packages/map_file/CMakeLists.txt?
If we use your solution as here: https://github.com/mikaelarguedas/perception_pcl/blob/fb67f90e94d1a00d371527862508e0df3707f22d/pcl_ros/CMakeLists.txt#L10-L22, should we be just stripping exactly these libs: https://github.com/mikaelarguedas/perception_pcl/blob/fb67f90e94d1a00d371527862508e0df3707f22d/pcl_ros/CMakeLists.txt#L16-L21 or also e.g. Qt5Core?

2) Packages that also use pcl_ros, e.g. https://github.com/CPFL/Autoware/blob/develop/ros/src/sensing/filters/packages/points_preprocessor/CMakeLists.txt. I assume since pcl_ros libraries link against all vtk libraries used by PCL, including the ones depending on Qt, now also our pkg links against Qt libraries.

Do we also need to strip these libraries https://github.com/mikaelarguedas/perception_pcl/blob/fb67f90e94d1a00d371527862508e0df3707f22d/pcl_ros/CMakeLists.txt#L16-L21 out of catkin_LIBRARIES variable?

I assume your fix above will not make it into kinetic, right?

Thx a lot, D.

mikaelarguedas, 2017-08-09:
Hi @dejanpan ,

> I assume your fix above will not make it into kinetic, right?

I think that fix was only necessary for Debian Stretch so didn't **need** to make it into kinetic. So assuming that you are testing on Ubuntu 16.04 and planning on releasing in kinetic you should not face that specific problem.

> I like the real solution by not linking against the library that you do not use.

Yeah conceptually I like it too, but the solution ends up being pretty ugly just because of how these libraries have been packaged upstream... too bad.

Regarding the `Qt5Core` libraries you mentioned:
without a build log and/or the time to reproduce it myself I'm going to give how I would go about it theoretically (maybe there is a better process that's just what I did for the `pcl_ros` issue)
The goal is to figure out why you face the problem in the first place (while it doesnt apprear in pcl_ros for example: see their [kinetic branch](https://github.com/ros-perception/perception_pcl/blob/kinetic-devel/pcl_ros/CMakeLists.txt)).
1) Isolate which one of your targets is trying to link against `Qt5Core`
2) Isolate which set of LIBRARIES contain that culprit library `Qt5Core` you try to link against (note that it can also be a target name like it was for the vtk targets in https://github.com/ros-perception/perception_pcl/pull/151)
3) Figure out if you actually need this library or not (it could be one of your dependency that needs it and doesn't export it properly)

3.a) If you don't need it, figure out why it is exposed to your target:
- if it's code you have control on: fix the underlying dependency
- if not: report the issue upstream on the appropriate repo

3.b) If you do need it: find which library taget you need to strip (using result of 1 and 2)

And then you should be good to go :wink:

closed 2020-05-29 by gbiggs