autowarefoundation/autoware #782: Can't launch the Runtime Manager in ubuntu 14.04

yulpun, 2017-08-23:
# Bug
## Expected Behavior
When I run the "./run" in the docker terminal, there was only one rosnode launched which named roscore,but I can't see the Runtime Manager window. Then I ran roscore in one terminal, 
and run $ cd $HOME/Autoware/ros/; source devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py
in another terminal.

The error message:
_Traceback (most recent call last):
  File "/root/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 84, in <module>
    from ndt_localizer.msg import ndt_stat
ImportError: No module named msg_

Hope response

amc-nu, 2017-08-29:
@yulpun 
Did you compile the code?
If the message files are missing, it might mean that these haven't been created.
Try following the https://github.com/CPFL/Autoware#how-to-build section.
Hope this helps.

closed 2017-10-30 by amc-nu

Tikitikitiki, 2018-07-10:
Try to follow the how to build part in this link to build: https://github.com/CPFL/Autoware/wiki/Source-Build