autowarefoundation/autoware #789: Build Error @develop branch

andoh104, 2017-09-02:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
No Error

## Actual Behavior
```
[ 93%] Built target ndt_mapping
/home/xxxxxx/Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/src/compress2png.cpp:3:41: fatal error: velodyne_pointcloud/rawdata.h: そのようなファイルやディレクトリはありません
 #include "velodyne_pointcloud/rawdata.h"
                                         ^
/home/xxxxxx/Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/src/recon2packet.cpp:2:41: fatal error: velodyne_pointcloud/rawdata.h: そのようなファイルやディレクトリはありません
 #include "velodyne_pointcloud/rawdata.h"
                                         ^
```

## Steps to Reproduce the Problem

  1.
`:~/Autoware/ros$ ./catkin_make_release
` 

## Specifications

  - Ubuntu version : 14.4
  - ROS Version: indigo
  - Autoware branch: develop

andoh104, 2017-09-02:
I think that this cause is as folows:

```
:~/Autoware$ git diff
diff --git a/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/CMakeLists.txt b/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/CMakeLists.txt
index 4f2e5f7a..7ff6f323 100644
--- a/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/CMakeLists.txt
+++ b/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/CMakeLists.txt
@@ -12,6 +12,7 @@ find_package(catkin REQUIRED COMPONENTS
   sensor_msgs
   velodyne_msgs
   pcl_conversions
+  velodyne_pointcloud
   )
 find_package( OpenCV REQUIRED )
 
diff --git a/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/package.xml b/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/package.xml
index a0b7fada..07b8ccab 100644
--- a/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/package.xml
+++ b/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_compression/package.xml
@@ -18,6 +18,7 @@
   <build_depend>sensor_msgs</build_depend>
   <build_depend>velodyne_msgs</build_depend>
   <build_depend>compression_msgs</build_depend>
+  <build_depend>velodyne_pointcloud</build_depend>
 
   <run_depend>roscpp</run_depend>
   <run_depend>rospy</run_depend>
```

dejanpan, 2017-09-02:
yup, that is clearly a bug as we use velodyne_pointcloud headers. Can you add `velodyne_pointcloud` also to `run_depend` tag and submit a PR please?

andoh104, 2017-09-02:
@dejanpan 
I can make a PR. but I can't test the velodyne_compression .
So, I will make a PR, please test it.

closed 2017-09-03 by yk-fujii