autowarefoundation/autoware #795: VTK Not Found on Ubuntu 16.04 LTS

maximus5684, 2017-09-06:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
CMake file for VTK in /usr/lib/cmake/vtk-6.2/VTKTargets.cmake points to /usr/bin/vtk. On systems with vtk6 installed, this does not exist.

## Actual Behavior
/usr/bin/vtk not found

## Steps to Reproduce the Problem

  1. Download Autoware.
  2. Initialize Catkin workspace.
  3. Run ./catkin_make_release

The following error is reported in the build output:
-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

## Specifications

  - Ubuntu version: 16.04.3 LTS
  - ROS Version: Kinetic
  - Autoware branch: master

## Possible solution:
`sudo update-alternatives --install /usr/bin/vtk vtk /usr/bin/vtk6 10

maximus5684, 2017-09-06:
Related to this, the file /usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so is also not found but both of these appear to be vtk6-related bugs. The above fixes the vtk dependency but the libvtkRenderingPythonTkWidgets.so only seems to exist in /usr/lib/python2.7/dist-packages/vtk. Creating a symlink with the following does the trick for this one:

`sudo ln -s /usr/lib/python2.7/dist-packages/vtk/libvtkRenderingPythonTkWidgets.x86_64-linux-gnu.so /usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so`

dejanpan, 2017-09-06:
@maximus5684 thx a lot. I am working on fixing Autoware packages ( https://github.com/CPFL/Autoware/issues/766)  so we can release them on the ROS build farm.

Do you expect this to get fixed with Ubuntu 16.04 updates?

On a related note, we have another issue with one of the pcl libraries which indirectly links in a whole swamp of VTK libraries: https://github.com/CPFL/Autoware/issues/773. 
Concrete details are given here: https://github.com/ros-perception/perception_pcl/pull/151.

Do you happen to have any experience with it?

maximus5684, 2017-09-08:
@dejanpan - According to the maintainer, both files referenced above which are called from VTKTargets.cmake are almost never required to be built against but they get pulled in from a monolithic CMake entry point (VTKTargets). Thus, he says, the warnings can be ignored. Here is the thread discussing this issue: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=826048

Based on the above, I think you can close this issue.

As for the Qt/VTK dependencies, I am not incredibly familiar with the chain but it looks like someone will likely have to go through the dependencies in each package that require Qt5 and figure out who requires it and why. Sorry I can't be of more help on this issue.

amc-nu, 2018-03-12:
https://stackoverflow.com/questions/37369369/compiling-pcl-1-7-on-ubuntu-16-04-errors-in-cmake-generated-makefile

kfunaoka, 2018-07-13:
Solved

closed 2018-07-13 by kfunaoka