autowarefoundation/autoware #798: Understanding Pure Pursuit Planner outputs

prarobo, 2017-09-12:
I am trying to understand the implementation of pure pursuit planner. I do not understand the relevance of some of the topics it is publishing. I am mainly looking at the following topics.
```
/ctrl_cmd [autoware_msgs/ControlCommandStamped]
/twist_raw [geometry_msgs/TwistStamped]
```
**Since the ```/ctrl_cmd``` topic provides a linear velocity and steering angle, how are the linear and angular velocity values coming from the ```/twist_raw``` being used?**

I was able to follow the data from these messages all the way to the vehicle sender node. In the vehicle_sender node, I see this block of code where a string of inputs is generated.

```
  std::ostringstream oss;
  oss << command_data.linear_x << ",";
  oss << command_data.angular_z << ",";
  oss << command_data.modeValue << ",";
  oss << command_data.gearValue << ",";
  oss << command_data.accellValue << ",";
  oss << command_data.brakeValue << ",";
  oss << command_data.steerValue << ",";
  oss << command_data.linear_velocity << ",";
  oss << command_data.steering_angle;
```

**Can someone explain the difference between __linear_x__ and __linear_velocity__? Similarly, what is the difference between __steerValue__ and __steering_angle__?**

Thanks.

alexandru-dinu, 2018-03-15:
We are also interested in this. Any help is appreciated!

siatwangmin, 2018-06-26:
This is calc from the ackeman car control model , with veloctiy and angular,  the culvature is determined, kappa = 1/R = w/v,  with ackeman car model, the steer_angle = arctan(wheel_base*kappa), for more informantion please see the video. https://www.youtube.com/watch?v=i6uBwudwA5o&list=PLcXdwRQiNMIoAOcJZvq5_c2xsoegDxu4N

closed 2019-03-01 by gbiggs