autowarefoundation/autoware #802: Build Error

Shuo-Xiong, 2017-09-19:
[ 13%] Built target g2o
[ 13%] Built target pcd_filter
Linking CXX executable /home/allen/Autoware/ros/devel/lib/object_map/laserscan2costmap
[ 13%] Built target laserscan2costmap
Linking CXX executable /home/allen/Autoware/ros/devel/lib/object_map/points2costmap
[ 13%] Built target points2costmap
Linking CXX static library /home/allen/Autoware/ros/devel/lib/liborb_slam2.a
[ 13%] Built target orb_slam2
make: *** [all] Error 2
Invoking "make -j56 -l56" failed

## Specifications

  - Ubuntu version:14.04
  - ROS Version:indigo
  - Autoware branch:

yk-fujii, 2017-09-20:
Thanks for the report.
Could you send the build log in full text?
The error log is not included.

Shuo-Xiong, 2017-09-20:
Thanks for the reply.
i updated the cmakeerror.log file, is the log file you want? or you need the contents in terminal windows?

Shuo-Xiong, 2017-09-20:
Thanks for the reply.
i updated the cmakeerror.log file, is the log file you want? or you need the contents in terminal windows?

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec422379412/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec422379412.dir/build.make CMakeFiles/cmTryCompileExec422379412.dir/build
make[1]: Entering directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec422379412.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec422379412.dir/CheckSymbolExists.c.o   -c /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec422379412
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec422379412.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec422379412.dir/CheckSymbolExists.c.o  -o cmTryCompileExec422379412 -rdynamic 
CMakeFiles/cmTryCompileExec422379412.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec422379412] Error 1
make: *** [cmTryCompileExec422379412/fast] Error 2

File /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec360664738/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec360664738.dir/build.make CMakeFiles/cmTryCompileExec360664738.dir/build
make[1]: Entering directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec360664738.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec360664738.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec360664738
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec360664738.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec360664738.dir/CheckFunctionExists.c.o  -o cmTryCompileExec360664738 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec360664738] Error 1
make: *** [cmTryCompileExec360664738/fast] Error 2

Determining if the Q_WS_WIN exist failed with the following output:
Change Dir: /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec3620934732/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3620934732.dir/build.make CMakeFiles/cmTryCompileExec3620934732.dir/build
make[1]: Entering directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building CXX object CMakeFiles/cmTryCompileExec3620934732.dir/CheckSymbolExists.cxx.o
/usr/bin/c++    -I/usr/include/qt4    -o CMakeFiles/cmTryCompileExec3620934732.dir/CheckSymbolExists.cxx.o -c /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx
/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx: In function ‘int main(int, char**)’:
/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:8:19: error: ‘Q_WS_WIN’ was not declared in this scope
   return ((int*)(&Q_WS_WIN))[argc];
                   ^
make[1]: Leaving directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
make[1]: *** [CMakeFiles/cmTryCompileExec3620934732.dir/CheckSymbolExists.cxx.o] Error 1
[... 55 lines cut ...]
/usr/bin/cmake -E cmake_progress_report /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building CXX object CMakeFiles/cmTryCompileExec2371585260.dir/CheckSymbolExists.cxx.o
/usr/bin/c++    -I/usr/include/qt4    -o CMakeFiles/cmTryCompileExec2371585260.dir/CheckSymbolExists.cxx.o -c /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx
/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx: In function ‘int main(int, char**)’:
/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:8:19: error: ‘Q_WS_MAC’ was not declared in this scope
   return ((int*)(&Q_WS_MAC))[argc];
                   ^
make[1]: *** [CMakeFiles/cmTryCompileExec2371585260.dir/CheckSymbolExists.cxx.o] Error 1
make[1]: Leaving directory `/home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp'
make: *** [cmTryCompileExec2371585260/fast] Error 2

File /home/allen/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_MAC
  return ((int*)(&Q_WS_MAC))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Shuo-Xiong, 2017-09-20:
Thanks a lot for the reply.?
you need the cmakeerror.log file in Autoware/ros/build folder or the build log in terminal windows?

Shuo-Xiong, 2017-09-20:
CMakeFiles/dist_transform.dir/nodes/dist_transform/dist_transform.cpp.o: In function `distanceTransform':
/home/allen/Autoware/ros/src/computing/perception/semantics/packages/object_map/nodes/dist_transform/dist_transform.cpp:57: undefined reference to `cv::distanceTransform(cv::_InputArray const&, cv::_OutputArray const&, int, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/allen/Autoware/ros/devel/lib/object_map/dist_transform] Error 1
make[1]: *** [computing/perception/semantics/packages/object_map/CMakeFiles/dist_transform.dir/all] Error 2

get_Depth.cpp:(.text._ZN2cv6StringC2ERKSs[_ZN2cv6StringC5ERKSs]+0x34): undefined reference to `cv::String::allocate(unsigned long)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/allen/Autoware/ros/devel/lib/data_preprocessor/get_Depth] Error 1
make[1]: *** [util/packages/data_preprocessor/CMakeFiles/get_Depth.dir/all] Error 2

/home/allen/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc: In member function ‘void ORB_SLAM2::Map::extractVocabulary(ORB_SLAM2::ORBVocabulary*)’:
/home/allen/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc:423:50: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<ORB_SLAM2::KeyFrame*>::size_type {aka long unsigned int}’ [-Wformat=]
  fprintf (stderr, "KF: %d\n", allKeyFrames.size());
                                                  ^
make[2]: *** [util/packages/kitti_pkg/kitti_player/CMakeFiles/kitti_player.dir/src/kitti_player.cpp.o] Error 1
make[1]: *** [util/packages/kitti_pkg/kitti_player/CMakeFiles/kitti_player.dir/all] Error 2

Shuo-Xiong, 2017-09-20:
Thanks a lot for the reply.
the text above is the error log in terminal

Fancy93, 2017-09-20:
I have the same issues with you,Can you tell me how you solved ?

yk-fujii, 2017-09-20:
Have you installed OpenCV?

The entity of cv :: distanceTransform is in libopencv_imgproc.so, 
but it looks be not linked in your environment. 

In addition, I was not reproduced this error in my environment(ubuntu 14.04/indigo).
However, since this package indirectly links OpenCV,  it may be bad depending on the os/ros/opencv version. 
I will fix a package to directly link OpenCV.

If you are in a hurry,  please comment out the part below.
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/semantics/packages/object_map/CMakeLists.txt#L50-L51

Shuo-Xiong, 2017-09-20:
Thanks a lot!
I have installed the OpenCV3.0
I solved the problem now
I removed the OpenCV3.0 and installed OpenCV2.4.13
also installed some dependencies below 

sudo apt-get install build-essential libgtk2.0-dev libjpeg-dev libtiff4-dev libjasper-dev libopenexr-dev cmake python-dev python-numpy python-tk libtbb-dev libeigen2-dev yasm libfaac-dev libopencore-amrnb-dev libopencore-amrwb-dev libtheora-dev libvorbis-dev libxvidcore-dev libx264-dev libqt4-dev libqt4-opengl-dev sphinx-common texlive-latex-extra libv4l-dev libdc1394-22-dev libavcodec-dev libavformat-dev libswscale-dev

yk-fujii, 2017-09-25:
Great！thanks!

closed 2017-09-25 by yk-fujii