autowarefoundation/autoware #810: Generic mailing list should never be a maintainer

130s, 2017-09-21:
Starting on Aug. 5th, mailing list subscribers have been receiving [notifications ROS buildfarm](https://groups.google.com/forum/#!topic/autoware/ApixQTtEUcQ). This is a sign that something is wrong and usually maintainers are asked to look into the issue.

#[Looks like](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=autoware) only source build has been successful.

And, for one of the packages in the devel repo, [user group's mailinglist email address is used as maintainer's email](https://github.com/CPFL/Autoware/blob/8351155ccd87d2d01c2f59f758d7fec4206e176b/ros/src/sensing/drivers/camera/packages/vectacam/package.xml#L7). Hence all the subscribers I believe are notified for every single build failure. I tried to open a PR for this but then had no idea what email I should use. Please fix this accordingly. Thanks.

dejanpan, 2017-09-21:
Ack. Will fix by disabling build on PRs. The problem is that this project
has 90+ projects and I can not release all of them at once.

130s, 2017-09-21:
I hear you.
Hope you'll have plenty of help at ROSCon :)

130s, 2017-10-06:
@amc-nu This seems to be your email address that needs fixed. Any extra cycle to spare on this (should be quick fix)?

dejanpan, 2017-10-09:
@130s I also fixed mail address for the maintainer. Thx for bringing this up.

closed 2017-10-09 by dejanpan