autowarefoundation/autoware #812: ./catkin_make_release Error

Fancy93, 2017-09-22:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
[ 12%] Generating Python msg __init__.py for orb_localizer
[ 12%] Built target orb_localizer_generate_messages_py
Scanning dependencies of target orb_vocabulary
In file included from /home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LocalMapping.h:26:0,
                 from /home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/System.h:34,
                 from /home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Viewer.h:28,
                 from /home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Tracking.h:30,
                 from /home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Tracking.cc:22:
/home/pan/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LoopClosing.h:34:46: fatal error: g2o/types/types_seven_dof_expmap.h: No such file or directory
 #include "g2o/types/types_seven_dof_expmap.h"
                                              ^
compilation terminated.
[ 12%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o
[ 12%] Built target orb_localizer_generate_messages_eus
Scanning dependencies of target freeglut_minimal
.
.
.

/home/pan/Autoware/ros/src/util/packages/data_preprocessor/nodes/get_Depth/get_Depth.cpp:170:14: warning: ‘h’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     for (int h; h<height; h++){
              ^
[ 14%] Linking CXX executable /home/pan/Autoware/ros/devel/lib/data_preprocessor/get_Depth
[ 14%] Built target get_Depth
[ 14%] Linking CXX executable /home/pan/Autoware/ros/devel/lib/data_preprocessor/get_PCD
[ 14%] Linking CXX static library /home/pan/Autoware/ros/devel/lib/libpangolin.a
[ 14%] Built target pangolin
[ 14%] Built target get_PCD
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

## Specifications

  - Ubuntu version: 14.04
  - ROS Version: indigo
  - Autoware branch: Autoware

sujiwo, 2017-09-22:
Sorry @Fancy93 , I couldn't reproduce error regarding to orb_localizer node. Are you sure it is in latest version of the branch? Could you please try this command:
`catkin_make --pkg orb_localizer`

Fancy93, 2017-09-26:
hi,@sujiwo  it was also failed,I compiled the g2o successfully,but the orb_localizer package cannot be compiled when I use the command  catkin_make --pkg orb_localizer or ./catkin_make_release .

Hierom, 2017-10-18:
Have same problem now.

Hierom, 2017-10-18:
Hello, @sujiwo 
I have a problem happened while orb_slam2.
I used 14.04 + opencv 3.3, and run the 
`catkin_make --pkg orb_localizer` too,
but not it has an error like this.
`
[ 87%] Built target pangolin
[ 87%] Built target orb_vocabulary
[ 87%] Built target orb_localizer_generate_messages_eus
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

Could you please help me for fix this? 
Thanks a lot.

wsung1, 2018-01-07:
@Fancy93 @sujiwo Hi, may I ask you whether this problem was solved?

sujiwo, 2018-01-07:
Please show full error message?

wsung1, 2018-01-07:
@sujiwo Nice to talk to you! Actually, my problem a bit differs with this. My problem has been posted as https://github.com/CPFL/Autoware/issues/1013. Please help!