autowarefoundation/autoware #815: Build Error @1.5.0

andoh104, 2017-09-24:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
No Error

## Actual Behavior
```
[  2%] Building CXX object util/packages/map_tools/CMakeFiles/pcd2csv.dir/nodes/pcd_converter/pcd2csv.cpp.o
/home/tandoh/Autoware/ros/src/util/packages/map_tools/nodes/map_extender/map_extender.cpp:49:45: fatal error: velodyne_pointcloud/point_types.h: そのようなファイルやディレクトリはありません
 #include <velodyne_pointcloud/point_types.h>
                                             ^
compilation terminated.
```

## Steps to Reproduce the Problem

`XXX:~/Autoware/ros$ ./catkin_make_release`

## Specifications

  - Ubuntu version:14.04
  - ROS Version:Indigo
  - Autoware branch:1.5.0 & master

andoh104, 2017-09-24:
I think this cause is as follows:
```
XXX:~/Autoware/ros$ git diff
diff --git a/ros/src/util/packages/map_tools/CMakeLists.txt b/ros/src/util/packages/map_tools/CMakeLists.txt
index 062509ef..e0484298 100644
--- a/ros/src/util/packages/map_tools/CMakeLists.txt
+++ b/ros/src/util/packages/map_tools/CMakeLists.txt
@@ -4,6 +4,7 @@ project(map_tools)
 find_package(catkin REQUIRED COMPONENTS
   pcl_ros
   pcl_conversions
+  velodyne_pointcloud
 #  runtime_manager
 )
 
diff --git a/ros/src/util/packages/map_tools/package.xml b/ros/src/util/packages/map_tools/package.xml
index e601d63b..319e9613 100644
--- a/ros/src/util/packages/map_tools/package.xml
+++ b/ros/src/util/packages/map_tools/package.xml
@@ -6,6 +6,7 @@
   <maintainer email="yuki@ertl.jp">kitsukawa</maintainer>
   <license>BSD</license>
   <buildtool_depend>catkin</buildtool_depend>
+  <build_depend>velodyne_pointcloud</build_depend>
   <export>
   </export>
 </package>

```
But after fixing this error, other errors still remain on the branch.

yk-fujii, 2017-09-25:
Thanks for the your report!
I will fix other issues together in 1.5.1 today.

closed 2017-09-26 by amc-nu