autowarefoundation/autoware #820: Unreachable code in lane_rule.cpp

histfuture, 2017-09-27:
Hi, I found a piece of strange code in lane_rule.cpp as the following shows:

line 123 - line 142
`
        bool crossroad = false;
	std::vector<size_t> start_indexes;
	std::vector<size_t> end_indexes;
	for (size_t i = 0; i < l.waypoints.size(); ++i) {
		vector_map::DTLane dtlane = lane_planner::vmap::create_vector_map_dtlane(l.waypoints[i].dtlane);
		if (i == 0) {
			crossroad = lane_planner::vmap::is_crossroad_dtlane(dtlane);
			continue;
		}
               //here, the code will not reach here forever.
		if (crossroad) {
			if (!lane_planner::vmap::is_crossroad_dtlane(dtlane)) {
				end_indexes.push_back(i - 1);
				crossroad = false;
			}
		} else {
			if (lane_planner::vmap::is_crossroad_dtlane(dtlane)) {
				start_indexes.push_back(i);
				crossroad = true;
			}
		}
	}
`

asimay, 2017-12-05:
why? if(i == 0) , here will redefine crossroad variable.

dejanpan, 2017-12-11:
@histfuture what @asimay says is correct. if you found an actual bug in behaviour please re-open.

closed 2017-12-11 by dejanpan