autowarefoundation/autoware #825: [Feature] Ray based ground filter

amc-nu, 2017-10-02:
# New Feature
To Provide a ground filter that doesn't require ring info (since it is not always available).

## Expected Result
VLP-16: https://drive.google.com/open?id=0BzYuVrO9pnh6SHdXRHFpSDNUVDA
HDL-32: https://drive.google.com/open?id=0BzYuVrO9pnh6MURaSlRZMFJzR3c
HDL-64: https://drive.google.com/open?id=0BzYuVrO9pnh6T0lsQ1lfaC0xRGs
In the videos the color
Red: Represents points classified as **not ground**
Blue: Points classifies as **ground**
White: PointCloud from the sensor

## Steps to Test
1. Play a Rosbag with LiDAR Data
2. Launch the Node from Runtime Manager in the *Sensing* tab -> *Points Preprocessor*(ray_ground_filter) or launch from the terminal using:
`$ roslaunch points_preprocessor ray_ground_filter`
3. The node will use as input the pointcloud message defined in `input_point_cloud` (default: `points_raw`)
4. The node will output the topics: `/points_ground` and `/points_no_ground`

The `sensor_height` and the `local_max_slope` and `global_max_slope` are essential parameters to the algorithm to work. These can be set using either the launch file or with the help of runtime manager (the app button). For more details about these parameters, please refer to the launch file or the descriptions in the UI.

## Specifications
Tested on
  - Ubuntu version: 14.04
  - ROS Version: 1.11.21
  - Autoware branch: *feature/ray_ground_filter*

amc-nu, 2017-10-02:
Also tested on
* Ubuntu 16.04
* ROS Version: Kinetic 1.12.7

closed 2017-10-06 by amc-nu

kargarisaac, 2018-11-12:
@amc-nu 
I cannot find the reference paper for this method. Is there any reference to read more about ray ground filter?

onatkeser, 2019-09-30:
same question with @kargarisaac , still no reference for ray ground filter?

onatkeser, 2019-09-30:
Also, is there any chance to understand (for the vehicle) whether there is a pothole (or gap or cliff etc.) or not with that ray ground filter. (In order not to fall basically) If not do you have any suggestions about that @amc-nu ?

nisailhan, 2021-06-24:
Same question with @onatkeser @kargarisaac. Is there any update @amc-nu?

amc-nu, 2021-06-24:
The algorithm is documented in the following paper:
```
@article{Abraham MONRROY CANO20202019TSP0005,
  title={An Open Multi-Sensor Fusion Toolbox for Autonomous Vehicles},
  author={Abraham MONRROY CANO and Eijiro TAKEUCHI and Shinpei KATO and Masato EDAHIRO},
  journal={IEICE Transactions on Fundamentals of Electronics, Communications and Computer Sciences},
  volume={E103.A},
  number={1},
  pages={252-264},
  year={2020},
  doi={10.1587/transfun.2019TSP0005}
}
```