autowarefoundation/autoware #840: Cannot Launch node of Type [kitti_player/kitti_player], can't locate node [kitti_player]...

rongliu1996, 2017-10-17:
roslaunch kitti_launch kittiplayer.launch failed with error: Cannot Launch node of Type [kitti_player/kitti_player], can't locate node [kitti_player] in package [kitti_player].  Understand that it is the same issue as the original question in issue #705 but keen to know if ROS Kinetic is the cause of this issue and how to resolve it.  Many thanks in advance!

## Reproduce steps
  1.  followed instruction in https://github.com/CPFL/Autoware/tree/master/ros/src/util/packages/kitti_pkg;
  2.  rebuilt Autoware ./catkin_make_release
  3.  ran $ source ./devel/setup.bash under Autoware/ros
  4.  ran $ roslaunch kitti_launch kittiplayer.launch  under Autoware/ros

## Specifications

  - Ubuntu version:  16.4.1 LTS
  - ROS Version:  Kinetic
  - Autoware branch: master

dejanpan, 2017-10-20:
@rongliu1996 yes, this node only gets compiled in indigo and jade versions: https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/kitti_pkg/kitti_player/CMakeLists.txt#L9

@amc-nu do you know, is there a reason for not that kitti_player would not compile on kinetic?

amc-nu, 2017-10-20:
@dejanpan I think it should work fine in Kinect. I’m not sure why it wasn’t included. @y-fujii do you know the reason?

yk-fujii, 2017-10-20:
Previously, kitti could not be compiled with kinetic.
It was resolved by the PR #653.
however, we forgot to delete the if sentence, and It has been neglected until now.

https://github.com/CPFL/Autoware/pull/653

amc-nu, 2017-10-20:
@yk-fujii thanks!
Checking further this node wasn’t updated to the new autoware_msgs. This would also be required.

rongliu1996, 2017-10-22:
Many thanks everybody.  Would you please advise an ETA for the new release?

mljack, 2017-10-26:
@amc-nu I'm new to autoware. Could you share how to fix the issue related to autoware_msgs you mentioned above?

I also hit this after remove the if and endif for indigo and jade version check. Seems that it's not related to autoware_msgs.
"/usr/bin/ld: cannot find -lQt5::Core"

amc-nu, 2017-10-29:
Please check #864

closed 2017-10-29 by amc-nu

shuzhenzhang, 2017-11-27:
hi,
have you resoved this problem?

amc-nu, 2017-11-27:
@shuzhenzhang Please check #864

shuzhenzhang, 2017-11-28:
@amc-nu 
Thanks for your reply, could you tell me how to do it?
I git clone master yesterday and compiled successfully.When I ran roslaunch kitti_launch kittiplayer.launch under Autoware/ros,the same error occurred.