autowarefoundation/autoware #845: [Bug] HDL-64e: Unable to open calibration file

aohsato, 2017-10-18:
I cannot run Velodyne HDL64e-S2 driver due to the incorrect calibration file.

# Bug
## Expected Behavior
```
SUMMARY
========

PARAMETERS
* /driver_nodelet/frame_id: velodyne
* /driver_nodelet/model: 64E_S2
* /driver_nodelet/pcap:
* /driver_nodelet/read_fast: False
* /driver_nodelet/read_once: False
* /driver_nodelet/repeat_delay: 0.0
* /driver_nodelet/rpm: 600.0
* /rosdistro: kinetic
* /rosversion: 1.12.7
* /velodyne_nodelet/calibration: /home/aohsato/wor...
* /velodyne_nodelet/max_range: 250.0
* /velodyne_nodelet/min_range: 2.0

NODES
/
driver_nodelet (nodelet/nodelet)
velodyne_nodelet (nodelet/nodelet)
velodyne_nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://Ocean.local:11311

core service [/rosout] found
YAML Exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ERROR] [1508333764.168792521]: Unable to open calibration file: /home/aohsato/workspace/projects/autoware/Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_pointcloud/params/64e_utexas.yaml
```

## Actual Behavior
No error occurs and a correct pointcloud is published.

## Steps to Reproduce the Problem

  1. Select the calibration file installed in Autoware.
```
Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_pointcloud/params/64e_utexas.yaml
Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_pointcloud/params/64e_s2.1-sztaki.yaml
```
[image]

  1. Run Sensing / Velodyne HDL-64e-S2

## Specifications

  - Ubuntu version: 16.04 LTS
  - ROS Version: Kinetic
  - Autoware branch: develop

dejanpan, 2017-10-20:
@aohsato just out of curiosity - is this happening because Velodyne updated their calibration files or why?

amc-nu, 2017-10-23:
@aohsato @dejanpan I also have the same problem on Indigo. 
To test, I cloned the original repo [ros-drivers/velodyne](https://github.com/ros-drivers/velodyne/tree/master)
I was able to decode the velodyne_packets using the `64e_utexas.yaml` using the following launch file
```
<launch>
  <arg name="pcap" default="" />
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/64e_utexas.yaml"/>
  <arg name="cloud_topic" default="points_raw"/>
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="model" value="64E"/>
    <arg name="pcap" value="$(arg pcap)"/>
  </include>
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
  </include>
</launch>
```
However, the same launch file will lead to a parsing error. Checking the code in Autoware, seems it was forked some time ago. A deeper analysis would be required to check what's the problem with the YAML parsing.
Another option is to replace again the code in Autoware with the one in the original repo, check the changes made in Autoware and merge them. (a quick glance allowed me to find, for instance the 64E_S3 was added in the Autoware's fork `velodyne_driver/src/driver/driver.cc`)

[image]

dejanpan, 2017-12-11:
@amc-nu do you know why Autoware decided to fork drivers rather than using original ones through submodules or https://github.com/dirk-thomas/vcstool (*.repos file)?

amc-nu, 2018-04-07:
Fixed on #1166 
@dejanpan afaik, the original maintainer added support for 64E_S3. Not included in the original repo.

closed 2018-04-07 by amc-nu