autowarefoundation/autoware #847: catkin_make --pkg orb_localizer error

Hierom, 2017-10-19:
Hi @sujiwo,

While I can't build up from `./catkin_make_release` (can't build orb_slam2), 
I saw some issues that you tell us can `catkin_make --pkg orb_localizer error` first.

But now, it still has an involking error.
`[ 87%] Built target pangolin
[ 87%] Built target orb_vocabulary
[ 87%] Built target orb_localizer_generate_messages_eus
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

Can you please tell me how to do next step for build up Autoware system?
My machine is Ubuntu 14.04 with OpenCV 3.3.0, Qt 5.3, but doesn't have Nvidia Cuda-package installed.

Thanks a lot in everything you have done in this repo.

Thank you.

sujiwo, 2017-10-19:
There are still some problems when compiling OpenCV 3.x. I'm still
investigating.

Hierom, 2017-10-19:
Okay, then which version is the best you suggest?

Thanks for telling me this.

sujiwo, 2017-10-19:
For ubuntu 14.04, it's best to stick with stock OpenCV 2.4.8 from
distribution.

Hierom, 2017-10-19:
But I used OpenCV 2.4.9 and failed in same problem.
Should I use above 2.4.10?

dejanpan, 2017-10-20:
@Hierom is this Ubuntu 16.04? develop branch? ROS kinetic?

Hierom, 2017-10-23:
@dejanpan, no, it's Ubuntu 14.04 with ROS indigo.
I still not test in OpenCV 2.4.10 yet, will report here after I test it.

Thanks all!

Hierom, 2017-10-25:
Dear @sujiwo and @dejanpan,
thanks for helping me to solve this issue.

For now, I use OpenCV 2.4.13, ROS indigo.
But while I recloned this repo and followed the How to Build way.
It still stopped while Building target orb_slam2 with just returned these in my terminal.

`**Linking CXX static library /home/chienrong/Github/Autoware/ros/devel/lib/liborb_slam2.a**
[ 12%] Built target orb_slam2
make: * [all] Error 2
Invoking "make -j4 -l4" failed`

Could you please tell me how to build this system up in the right way.

Many thanks.

sujiwo, 2017-10-26:
Can you please give me full output of the catkin_make? like:

catkin_make --pkg orb_localizer > compile_out.log 2>&1

Hierom, 2017-10-26:
Hi @sujiwo thanks for helping me.

The following is output in log file.
 
-- Using CATKIN_DEVEL_PREFIX: /home/chienrong/Github/Autoware/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /home/chienrong/miniconda3/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/chienrong/Github/Autoware/ros/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 100 packages in topological order:
-- ~~  - filters
-- ~~  - glviewer
-- ~~  - gnss
-- ~~  - graph_tools
-- ~~  - hexacam
-- ~~  - kitti_launch
-- ~~  - libdpm_ttic
-- ~~  - librcnn
-- ~~  - marker_downsampler
-- ~~  - ndt_tku
-- ~~  - nmea_navsat
-- ~~  - op_utility
-- ~~  - op_planner
-- ~~  - op_simu
-- ~~  - pc2_downsampler
-- ~~  - registration
-- ~~  - requirements_version_checker
-- ~~  - javad_navsat_driver
-- ~~  - sample_data
-- ~~  - autoware_msgs
-- ~~  - libdpm_ocv
-- ~~  - points2polygon
-- ~~  - tablet_socket_msgs
-- ~~  - udon_socket
-- ~~  - vectacam
-- ~~  - vector_map_msgs
-- ~~  - velodyne (metapackage)
-- ~~  - velodyne_msgs
-- ~~  - viewers
-- ~~  - autoware_connector
-- ~~  - custom_msgs
-- ~~  - euc
-- ~~  - fake_drivers
-- ~~  - fastvirtualscan
-- ~~  - fusion
-- ~~  - garmin
-- ~~  - hokuyo
-- ~~  - kitti_box_publisher
-- ~~  - kvaser
-- ~~  - lane_detector
-- ~~  - laser_scan_converter
-- ~~  - obj_db
-- ~~  - point_cloud_converter
-- ~~  - pos_db
-- ~~  - rosinterface
-- ~~  - calibration_camera_lidar
-- ~~  - points2image
-- ~~  - image_segmenter
-- ~~  - kf
-- ~~  - cv_tracker
-- ~~  - runtime_manager
[... 661 lines cut ...]
ImagePreprocessor.cpp:(.text+0x16f0): undefined reference to `cv::LUT(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx] Error 1
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/imgprocx.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target orb_localizer_generate_messages_cpp
Scanning dependencies of target orb_localizer_generate_messages_lisp
[ 75%] [ 75%] Generating Lisp code from orb_localizer/debug.msg
Generating C++ code from orb_localizer/debug.msg
[ 75%] Built target orb_localizer_generate_messages_lisp
[ 75%] Built target orb_localizer_generate_messages_cpp
[ 75%] Built target orb_vocabulary
make: *** [all] Error 2
Base path: /home/chienrong/Github/Autoware/ros
Source space: /home/chienrong/Github/Autoware/ros/src
Build space: /home/chienrong/Github/Autoware/ros/build
Devel space: /home/chienrong/Github/Autoware/ros/devel
Install space: /home/chienrong/Github/Autoware/ros/install
####
#### Running command: "cmake /home/chienrong/Github/Autoware/ros/src -DCATKIN_DEVEL_PREFIX=/home/chienrong/Github/Autoware/ros/devel -DCMAKE_INSTALL_PREFIX=/home/chienrong/Github/Autoware/ros/install -G Unix Makefiles" in "/home/chienrong/Github/Autoware/ros/build"
####
####
#### Running command: "make -j4 -l4" in "/home/chienrong/Github/Autoware/ros/build/computing/perception/localization/packages/orb_localizer"
####
Invoking "make -j4 -l4" failed

sujiwo, 2017-10-26:
It looks like catkin was unable to find OpenCV library; did you install it
in non-standard path? If so, try adding it in CMAKE_PREFIX_PATH environment
variable.
For example, if OpenCV is in /usr/local/opencv2

export CMAKE_PREFIX_PATH=/usr/local/opencv2:$CMAKE_PREFIX_PATH
catkin_make --pkg orb_localizer

Hierom, 2017-10-26:
@sujiwo  Many thanks! I change my CMAKE_PREFIX_PATH to /use/local/include/opencv2
And the following was output in log.

[  0%] Built target DBoW2
[ 25%] Built target pangolin
Linking CXX executable /home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx
[ 62%] Built target g2o
[ 62%] Built target std_msgs_generate_messages_py
[ 62%] Built target std_msgs_generate_messages_cpp
[ 62%] Built target std_msgs_generate_messages_lisp
[ 62%] Built target _orb_localizer_generate_messages_check_deps_debug
[ 75%] Built target orb_slam2
[ 75%] Built target orb_vocabulary
[ 75%] [ 75%] Built target orb_localizer_generate_messages_py
Built target orb_localizer_generate_messages_cpp
[ 75%] Built target orb_localizer_generate_messages_lisp
[ 87%] Built target orb_localizer_generate_messages_eus
[ 87%] Built target orb_mapping
[ 87%] Built target orb_mapping_offline
[ 87%] Built target orb_matching
[ 87%] Built target dumpmap
Scanning dependencies of target orb_localizer_generate_messages
[ 87%] Built target orb_localizer_generate_messages
[ 87%] Built target map_publisher
Scanning dependencies of target orb_matching_pf
CMakeFiles/imgprocx.dir/nodes/imgprocx/imgprocx.cpp.o: In function `main':
imgprocx.cpp:(.text.startup+0x9a6): undefined reference to `cv::imread(cv::String const&, int)'
imgprocx.cpp:(.text.startup+0x9ae): undefined reference to `cv::String::deallocate()'
imgprocx.cpp:(.text.startup+0xaec): undefined reference to `cv::String::allocate(unsigned long)'
imgprocx.cpp:(.text.startup+0xc08): undefined reference to `cv::String::deallocate()'
CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o: In function `ImagePreprocessor::setGamma(cv::Mat&, float, bool)':
ImagePreprocessor.cpp:(.text+0xe36): undefined reference to `cv::LUT(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&)'
CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o: In function `ImagePreprocessor::autoAdjustGammaRGB(cv::Mat&, std::vector<cv::Mat, std::allocator<cv::Mat> >&, cv::Mat)':
ImagePreprocessor.cpp:(.text+0x16f0): undefined reference to `cv::LUT(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx] Error 1
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/imgprocx.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] [100%] [100%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/nodes/orb_matching/FusionOdometry.cpp.o
Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/nodes/orb_matching/Matcher.cpp.o
Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/nodes/orb_matching/orb_matching_pf.cpp.o
[100%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/nodes/common.cpp.o
/home/chienrong/Github/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.cpp: In function ‘tf::Transform rotateAxes(const tf::Transform&, float, float, float)’:
/home/chienrong/Github/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.cpp:40:55: warning: ‘tf::Quaternion::Quaternion(const tfScalar&, const tfScalar&, const tfScalar&)’ is deprecated (declared at /opt/ros/indigo/include/tf/LinearMath/Quaternion.h:51) [-Wdeprecated-declarations]
  rotation.setRotation (tf::Quaternion(roll, pitch, yaw).normalize());
                                                       ^
/home/chienrong/Github/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.cpp: In static member function ‘static tf::Transform Matcher::localizeByReference(const tf::Transform&, const tf::Transform&, const tf::Transform&, const tf::Transform&, const tf::Transform&)’:
/home/chienrong/Github/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.cpp:408:57: warning: ‘tf::Quaternion::Quaternion(const tfScalar&, const tfScalar&, const tfScalar&)’ is deprecated (declared at /opt/ros/indigo/include/tf/LinearMath/Quaternion.h:51) [-Wdeprecated-declarations]
  flipAxes.setRotation (tf::Quaternion(M_PI/2, 0, -M_PI/2).normalize());
                                                         ^
Linking CXX executable /home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/orb_matching_pf
[100%] Built target orb_matching_pf
make: *** [all] Error 2
Base path: /home/chienrong/Github/Autoware/ros
Source space: /home/chienrong/Github/Autoware/ros/src
Build space: /home/chienrong/Github/Autoware/ros/build
Devel space: /home/chienrong/Github/Autoware/ros/devel
Install space: /home/chienrong/Github/Autoware/ros/install
####
#### Running command: "make cmake_check_build_system" in "/home/chienrong/Github/Autoware/ros/build"
####
####
#### Running command: "make -j4 -l4" in "/home/chienrong/Github/Autoware/ros/build/computing/perception/localization/packages/orb_localizer"
####
Invoking "make -j4 -l4" failed

This was the second time catkin_make log, but first time was stopped in 75%, 
but now can move on to 100% but stopped by orb_matching_pf.
Does it make sense for trying catkin_make several times to make some target successfully? Or it just passing it?

Thanks

sujiwo, 2017-10-26:
Try set CMAKE_PREFIX_PATH to /usr/local
And sorry, trying catkin_make several times won't fix problem.

Hierom, 2017-10-26:
@sujiwo I changed my PATH as you said to /usr/local.

But still stopped at the same place.

[  0%] Built target DBoW2
[ 25%] Built target pangolin
Linking CXX executable /home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx
[ 62%] Built target g2o
[ 62%] Built target std_msgs_generate_messages_py
[ 62%] Built target std_msgs_generate_messages_cpp
[ 62%] Built target std_msgs_generate_messages_lisp
[ 62%] Built target orb_vocabulary
[ 62%] Built target _orb_localizer_generate_messages_check_deps_debug
[ 75%] Built target orb_slam2
[ 75%] Built target orb_localizer_generate_messages_cpp
[ 75%] Built target orb_localizer_generate_messages_py
[ 75%] Built target orb_localizer_generate_messages_lisp
[ 87%] Built target orb_localizer_generate_messages_eus
[ 87%] [ 87%] Built target orb_mapping_offline
Built target orb_mapping
[ 87%] Built target orb_matching
[ 87%] Built target dumpmap
[ 87%] Built target orb_localizer_generate_messages
[100%] Built target orb_matching_pf
[100%] Built target map_publisher
CMakeFiles/imgprocx.dir/nodes/imgprocx/imgprocx.cpp.o: In function `main':
imgprocx.cpp:(.text.startup+0x9a6): undefined reference to `cv::imread(cv::String const&, int)'
imgprocx.cpp:(.text.startup+0x9ae): undefined reference to `cv::String::deallocate()'
imgprocx.cpp:(.text.startup+0xaec): undefined reference to `cv::String::allocate(unsigned long)'
imgprocx.cpp:(.text.startup+0xc08): undefined reference to `cv::String::deallocate()'
CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o: In function `ImagePreprocessor::setGamma(cv::Mat&, float, bool)':
ImagePreprocessor.cpp:(.text+0xe36): undefined reference to `cv::LUT(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&)'
CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o: In function `ImagePreprocessor::autoAdjustGammaRGB(cv::Mat&, std::vector<cv::Mat, std::allocator<cv::Mat> >&, cv::Mat)':
ImagePreprocessor.cpp:(.text+0x16f0): undefined reference to `cv::LUT(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx] Error 1
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/imgprocx.dir/all] Error 2
make: *** [all] Error 2
Base path: /home/chienrong/Github/Autoware/ros
Source space: /home/chienrong/Github/Autoware/ros/src
Build space: /home/chienrong/Github/Autoware/ros/build
Devel space: /home/chienrong/Github/Autoware/ros/devel
Install space: /home/chienrong/Github/Autoware/ros/install
####
#### Running command: "make cmake_check_build_system" in "/home/chienrong/Github/Autoware/ros/build"
####
####
#### Running command: "make -j4 -l4" in "/home/chienrong/Github/Autoware/ros/build/computing/perception/localization/packages/orb_localizer"
####
Invoking "make -j4 -l4" failed

sujiwo, 2017-10-26:
May I see output of this command ?

locate libopencv_core.so

sujiwo, 2017-10-26:
And this one:

locate libopencv_highgui.so

2017-10-26 12:08 GMT+09:00 Adi Sujiwo <sujiwo@gmail.com>:

Hierom, 2017-10-26:
@sujiwo 
Here it is,
chienrong@chienrong:~/Github/Autoware/ros$ locate libopencv_core.so
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_core.so
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_core.so.3.1
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_core.so.3.1.0
/home/chienrong/miniconda3/lib/libopencv_core.so
/home/chienrong/miniconda3/lib/libopencv_core.so.3.1
/home/chienrong/miniconda3/lib/libopencv_core.so.3.1.0
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_core.so
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_core.so.3.1
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_core.so.3.1.0
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_core.so
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_core.so.3.1
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_core.so.3.1.0
/usr/local/lib/libopencv_core.so

And
chienrong@chienrong:~/Github/Autoware/ros$ locate libopencv_highgui.so
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_highgui.so
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_highgui.so.3.1
/home/chienrong/miniconda3/envs/carnd-term1/lib/libopencv_highgui.so.3.1.0
/home/chienrong/miniconda3/lib/libopencv_highgui.so
/home/chienrong/miniconda3/lib/libopencv_highgui.so.3.1
/home/chienrong/miniconda3/lib/libopencv_highgui.so.3.1.0
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_highgui.so
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_highgui.so.3.1
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py35_0/lib/libopencv_highgui.so.3.1.0
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_highgui.so
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_highgui.so.3.1
/home/chienrong/miniconda3/pkgs/opencv3-3.1.0-py36_0/lib/libopencv_highgui.so.3.1.0
/usr/local/lib/libopencv_highgui.so

Hierom, 2017-10-26:
Does it mean I set the wrong PATH?
It should be /usr/local/lib?

sujiwo, 2017-10-26:
Let's keep using /usr/local as CMAKE_PREFIX_PATH.
Could you send me the file:

build/orb_localizer/CMakeFiles/imgprocx.dir/link.txt

should be under the directory in which you ran catkin_make

Hierom, 2017-10-26:
@sujiwo Sorry for late replying.

I didn't find the link.txt under my Autoware/ros/build directory, but found it in 

/Autoware/ros/build/computing/perception/localization/packages/orb_localizer/CMakeFiles/imgprocx.dir/

Here it is, and thank you so much.

/usr/bin/c++    -std=c++11 -fPIC -O3 -DNDEBUG    CMakeFiles/imgprocx.dir/nodes/imgprocx/imgprocx.cpp.o CMakeFiles/imgprocx.dir/nodes/imgprocx/ImagePreprocessor.cpp.o  -o /home/chienrong/Github/Autoware/ros/devel/lib/orb_localizer/imgprocx -rdynamic /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/libimage_transport.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /opt/ros/indigo/lib/libroscpp.so -lboost_signals /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libroslib.so /opt/ros/indigo/lib/librospack.so -lpython2.7 -lboost_filesystem -lboost_program_options -ltinyxml /opt/ros/indigo/lib/libcv_bridge.so /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.8 /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge -Wl,-rpath,/opt/ros/indigo/lib

Hierom, 2017-10-27:
@sujiwo Does it mean I should install 2.4.8 instead?

sujiwo, 2017-10-27:
I've confirmed that OpenCV 2.4.8 compiles cleanly with orb_slam. There may
be other problems, though.

Hierom, 2017-10-28:
@sujiwo Thanks a lot, let me try for OpenCV 2.4.8 later, by the way, you mentioned me that it should be installed "distributed", what does it mean?

Not like the common way download the install package and build it?

Thanks

closed 2017-11-22 by Hierom

wsung1, 2018-01-07:
@sujiwo Dear sujiwon, thanks for your support. I'm a newbie here, so please bear with me. According to talks here, you asked Hierom to do 'catkin_make --pkg orb_localizer' instead of './catkin_make_release'. Does it mean 'catkin_make --pkg orb_localizer' could substitute for './catkin_make_release'? In my case,  'catkin_make --pkg orb_localizer' resulted in no problem; yet './catkin_make_release' was invoking error.

sujiwo, 2018-01-07:
catkin_make --pkg orb_localizer compiles only this package, not the whole
Autoware workspace as ./catkin_make_release does. So when the former
results in no error but the latter does, I think the error does not lie in
orb_localizer.

wsung1, 2018-01-07:
@sujiwo Thanks for your quick reply, sujiwo. Then, can I ask you what can be the problem? FYI, I have fully followed the instruction (Autoware_UsersManual_v1.1.pdf) in terms of OpenCV version and something like that.