autowarefoundation/autoware #850: error: XDG_RUNTIME_DIR not set in the environment while using docker

cvcore, 2017-10-24:
I followed the instruction in `docker/README.md` to install Autoware docker image. However after the installation, `sh run.sh` complains about `error: XDG_RUNTIME_DIR not set in the environment.` and the docker image failed to start.

Original output:
```
sh run.sh
No protocol specified

** (gnome-terminal:1): WARNING **: Could not open X display
No protocol specified
error: XDG_RUNTIME_DIR not set in the environment.
Failed to parse arguments: Cannot open display: 
```
My desktop setup is lightdm + i3wm on Archlinux, with environment variables correctly setup as (echoing them gives me):
`DISPLAY=:0`

chenyang-08s, 2017-11-06:
Same problem :C

dejanpan, 2017-11-10:
@messi49 did you see this issues?

@cvcore I am fairly certain that no-one is running this combination lightdm + i3wm on Archlinux. Also not saying that it is a culprit but just no-one has tried it out.

cvcore, 2017-11-10:
@dejanpan Agreed on your second point. Disagreed the first. AFAIK running i3wm with lightdm as display manager somewhat common in i3 communities. But I guess this is off-topic for this issue so I prefer not discuss further.

current status of my issue:
I managed to get the autoware docker image up and running by first removing the gnome-terminal in dockerfile, installing glx-gears to debug the issue and running gnome-terminal again. Right now I'm not certain what'd be the cause of this issue, but I guess it's something related to X authentication.

@chenyang-08s https://github.com/CPFL/Autoware/issues/612 might be worth trying. But it's not working in my case.

dejanpan, 2017-11-11:
@cvcore sorry, I misspoke I meant to say that  that no-one in _Autoware community that I am aware of_ is running combination of lightdm + i3wm on Archlinux. Sorry for that.

closed 2020-05-29 by gbiggs