autowarefoundation/autoware #853: [Bug] NDT_Localizer CMake error when CUDA is not installed

amc-nu, 2017-10-25:
# Bug
Cannot compile `ndt_localizer` without CUDA installed on `develop` branch.

## Steps to Reproduce the Problem
In a system with no GPU or CUDA not installed.
  1. ./catkin_make release

```
-- Could not find the required component 'ndt_gpu'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "ndt_gpu" with any
  of the following names:

    ndt_gpuConfig.cmake
    ndt_gpu-config.cmake

  Add the installation prefix of "ndt_gpu" to CMAKE_PREFIX_PATH or set
  "ndt_gpu_DIR" to a directory containing one of the above files.  If
  "ndt_gpu" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  computing/perception/localization/packages/ndt_localizer/CMakeLists.txt:38 (find_package)

```

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: 1.12.7
  - Autoware branch: **develop**

## Analysis
ndt_gpu library checks if CUDA is installed on:
https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeLists.txt#L10

However, `ndt_localizer` package adds ndt_gpu without checking CUDA on:
https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/packages/ndt_localizer/CMakeLists.txt#L12

## Suggested solution
Change:
`SET(FAST_PCL_PACKAGES filters registration ndt_gpu)` to 
`SET(FAST_PCL_PACKAGES filters registration)`

and change the block starting on line [23](https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/packages/ndt_localizer/CMakeLists.txt#L24) to
```
if(CUDA_FOUND)
    add_definitions(-DCUDA_FOUND)
    list(APPEND FAST_PCL_PACKAGES ndt_gpu)
endif()
```

kitsukawa, 2017-10-25:
@amc-nu Thank you for your report and suggestion. I fixed this as you suggested in #854 . Can you check?

weisongwen, 2017-10-25:
@kitsukawa

weisongwen, 2017-10-25:
@kitsukawa 
Hi,  I encountered an error when I compile the newest Autoware version (develop branch) as following. But I have already install the GPU as I do the deep learning before.

# Bug
**/home/wenws/Autoware-develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp:117:8: error: ‘gpu’ does not name a type
 static gpu::GNormalDistributionsTransform gpu_ndt;
        ^
/home/wenws/Autoware-develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp: In function ‘void param_callback(const ConstPtr&)’:
/home/wenws/Autoware-develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp:258:7: error: ‘gpu_ndt’ was not declared in this scope
       gpu_ndt.setResolution(ndt_res);
       ^
/home/wenws/Autoware-develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp:274:7: error: ‘gpu_ndt’ was not declared in this scope
       gpu_ndt.setStepSize(step_size);**

## Specifications

  - Ubuntu version: Ubuntu 14.04
  - ROS Version: jade
  - Autoware branch: latest develop branch

#Error pictured as following:
[image]

amc-nu, 2017-10-27:
@kitsukawa I checked your PR, thanks.
However, I found another problem.

Please check _if_ block at line [80](https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/packages/ndt_localizer/CMakeLists.txt#L76) of `ndt_localizer` package.

*If you have a newer version of PCL* (>=1.7.2), you define `USE_FAST_PCL`.
This flag is used in `ndt_mapping.cpp` in line [61](https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_mapping/ndt_mapping.cpp#L61).
BUT, if you DON'T have CUDA neither the `voxel_grid.h` nor the `ndt.h` will be included. Causing an error in compilation.
```
error: ‘NormalDistributionsTransform’ in namespace ‘pcl’ does not name a template type
error: ‘ndt’ was not declared in this scope
```
Please check some updates I did.

closed 2017-10-27 by amc-nu

LAsurvivor, 2020-05-19:
I've installed the CUDA 9.0 on Ubuntu 16.04 ROS kinetic, but I encountered a similar problem as below:

`Starting >>> sick_ldmrs_laser
--- stderr: lidar_localizer                                                                   
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** visualization features related to openni2 will be disabled
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "ndt_gpu" with any
  of the following names:

    ndt_gpuConfig.cmake
    ndt_gpu-config.cmake

  Add the installation prefix of "ndt_gpu" to CMAKE_PREFIX_PATH or set
  "ndt_gpu_DIR" to a directory containing one of the above files.  If
  "ndt_gpu" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "ndt_gpu" with any
  of the following names:

    ndt_gpuConfig.cmake
    ndt_gpu-config.cmake

  Add the installation prefix of "ndt_gpu" to CMAKE_PREFIX_PATH or set
  "ndt_gpu_DIR" to a directory containing one of the above files.  If
  "ndt_gpu" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:40 (find_package)

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< lidar_localizer	[ Exited with code 2 ]
`

I'm compiling the autoware 1.12.0 from source using 
`$ AUTOWARE_COMPILE_WITH_CUDA=1 colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

How can I solve the problem?