autowarefoundation/autoware #855: px2 installation script bug.

asimay, 2017-10-26:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
in PX2 setup instructions, the step:` git checkout -b aarch64 origin/feature/aarch64`

nvidia@nvidia:~/Autoware$ git checkout -b aarch64 origin/feature/aarch64
fatal: Cannot update paths and switch to branch 'aarch64' at the same time.
Did you intend to checkout 'origin/feature/aarch64' which can not be resolved as commit?

it report fatal error....

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

yk-fujii, 2017-10-26:
Ah, I forgot to fix script;(
The feature/aarch64 branch have already merged to master.
Please ignore it line and compile with master branch.

asimay, 2017-10-26:
wow. thank you very much for so prompt answer.
I skip it, and build on a clean PX2, but still got much compiling error.

closed 2017-10-26 by dejanpan