autowarefoundation/autoware #857: px2 build & link error.

asimay, 2017-10-26:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
On a clean PX2, according to instruction of PX2 setup, when build:

[ 71%] Linking CXX executable /home/nvidia/Autoware/ros/devel/lib/freespace_planner/astar_navi
[ 71%] Built target astar_navi
[ 71%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/nodes/common.cpp.o
Scanning dependencies of target dumpmap
[ 71%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/dumpmap.dir/nodes/dumpmap/dumpmap.cc.o
[ 71%] Linking CXX executable /home/nvidia/Autoware/ros/devel/lib/orb_localizer/orb_mapping_offline
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/build.make:339: recipe for target '/home/nvidia/Autoware/ros/devel/lib/orb_localizer/orb_mapping_offline' failed
make[2]: *** [/home/nvidia/Autoware/ros/devel/lib/orb_localizer/orb_mapping_offline] Error 1
CMakeFiles/Makefile2:17024: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/all' failed
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 71%] Linking CXX executable /home/nvidia/Autoware/ros/devel/lib/orb_localizer/dumpmap
[ 71%] Built target dumpmap
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

yk-fujii, 2017-10-26:
@asimay 
It's a OpenGL compatibility issue.

It is written at the bottom of the file below.
https://github.com/CPFL/Autoware/blob/master/docs/en/installation_with_drivepx2.md

asimay, 2017-10-26:
ok. thank you very much . i find the cause. I exported the variance yesterday, but today not export, caused this issue.
thanks!

asimay, 2017-10-26:
i closed it

closed 2017-10-26 by asimay