autowarefoundation/autoware #858: [Bug] localization/packages/orb_localizer/src/Map.cc:423:50: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<ORB_SLAM2::KeyFrame*>::size_type {aka long unsigned int}’ [-Wformat=]   fprintf (stderr, "KF: %d\n", allKeyFrames.size());

jinfagang, 2017-10-26:
Build error in `Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc`
```
/home/jintian/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc:423:50: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<ORB_SLAM2::KeyFrame*>::size_type {aka long unsigned int}’ [-Wformat=]
  fprintf (stderr, "KF: %d\n", allKeyFrames.size());

```

dejanpan, 2017-11-09:
@jinfagang Please provide this information, before we can not help you:
```
# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:
```

aztinki, 2017-11-15:
#Bug 
## Autoware Build fails for ORB Localization
**Problem :** 
I was able to run Autoware smooothly, until I started messing around with the Point Cloud Library. I uninstalled and reinstalled PCL from source. The error I get mentions about an undefined reference, I think it is because I installed PCL from source. I am not sure what to do next. Any help will be much appreciated.
**Output of ./catkin_make_release**
Base path: /scratch/agtxti/Autoware/ros
Source space: /scratch/agtxti/Autoware/ros/src
Build space: /scratch/agtxti/Autoware/ros/build
Devel space: /scratch/agtxti/Autoware/ros/devel
Install space: /scratch/agtxti/Autoware/ros/install
####
#### Running command: "cmake /scratch/agtxti/Autoware/ros/src -DCATKIN_DEVEL_PREFIX=/scratch/agtxti/Autoware/ros/devel -DCMAKE_INSTALL_PREFIX=/scratch/agtxti/Autoware/ros/install -G Unix Makefiles" in "/scratch/agtxti/Autoware/ros/build"
####
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /scratch/agtxti/Autoware/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /scratch/agtxti/Autoware/ros/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 100 packages in topological order:
-- ~~  - filters
-- ~~  - glviewer
-- ~~  - gnss
-- ~~  - graph_tools
-- ~~  - hexacam
-- ~~  - kitti_launch
-- ~~  - libdpm_ttic
-- ~~  - librcnn
-- ~~  - marker_downsampler
-- ~~  - ndt_tku
-- ~~  - nmea_navsat
-- ~~  - op_utility
-- ~~  - op_planner
-- ~~  - op_simu
-- ~~  - pc2_downsampler
-- ~~  - registration
-- ~~  - requirements_version_checker
-- ~~  - javad_navsat_driver
-- ~~  - sample_data
-- ~~  - autoware_msgs
-- ~~  - libdpm_ocv
-- ~~  - points2polygon
[... 2827 lines cut ...]
[ 67%] Built target state_machine
Scanning dependencies of target orb_mapping
[ 67%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping.dir/nodes/orb_mapping/orb_mapping.cpp.o
Scanning dependencies of target orb_mapping_offline
Scanning dependencies of target orb_localizer_generate_messages_lisp
[ 67%] [ 67%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/nodes/orb_mapping/orb_mapping_offline.cpp.o
Generating Lisp code from orb_localizer/debug.msg
[ 67%] Built target orb_localizer_generate_messages_lisp
[ 68%] Scanning dependencies of target orb_localizer_generate_messages_py
Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/nodes/common.cpp.o
[ 68%] Generating Python from MSG orb_localizer/debug
[ 68%] Generating Python msg __init__.py for orb_localizer
[ 68%] Built target orb_localizer_generate_messages_py
[ 68%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping.dir/nodes/common.cpp.o
Scanning dependencies of target orb_localizer_generate_messages_cpp
[ 68%] Generating C++ code from orb_localizer/debug.msg
[ 68%] Built target orb_localizer_generate_messages_cpp
Scanning dependencies of target orb_localizer_generate_messages_eus
[ 68%] Generating EusLisp code from orb_localizer/debug.msg
[ 68%] Generating EusLisp manifest code for orb_localizer
Scanning dependencies of target op_simulator_perception
[ 68%] Built target orb_localizer_generate_messages_eus
Scanning dependencies of target map_publisher
[ 69%] Building CXX object computing/planning/motion/packages/op_simulator_perception/CMakeFiles/op_simulator_perception.dir/nodes/OpenPlannerSimulatorPerception.cpp.o
[ 69%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/map_publisher.dir/nodes/visualization

aztinki, 2017-11-16:
Solved it by reinstalling everything!

closed 2017-11-17 by amc-nu