autowarefoundation/autoware #861: [question] about do real car demo.

asimay, 2017-10-27:
I'm testing your great package on PX2 platform.
I wonder to know following cases:
1. If I  have a VLP-16, I don't have 3D HD map,  that means there is no vector map, just the 3D laser points. In this case, can I use your package to do localize or do NDT matching?
2.  Do you have any special way to produce the HD map by yourselves? is there any document or instruction from you?

Thank you very much! I really appreciate your help.

asimay, 2017-10-28:
The rosbag file is only about 91s. too short for testing. and after load the localization script, it is hard to localize the car's position.

amc-nu, 2017-10-29:
Please have a look at:
https://maptools.tier4.jp/

or you can contact a professional to create the HDMap for you.

We get ours from https://www.aisantec.co.jp/english/

closed 2017-10-29 by amc-nu