autowarefoundation/autoware #862: Compilation crash on orb_localizer package : addVertex function in g2o thirdpary library

GuillaumeHauss, 2017-10-27:
# Bug 
Autoware compilation crashes on orb_localizer package : addVertex function undefined

## Actual Behavior
When executing ./catkin_make_release to compile Autoware, the command crashes when compiling orb_localizer package -- more specifically these three sub nodes : 
`orb_matching`
`orb_matching_pf`
`orb_mapping_offline` 

The log trace mentions multiple undefined references to the following function: `g2o::OptimizableGraph::addVertex(g2o::HyperGraph::Vertex*, g2o::HyperGraph::Data*)`

At the beginning, I thought it was because g2o was not properly linked, but I checked out the makefile, and it clearly adds Thirdparty/g2o as dependancy.
`# For building g2o`
`add_subdirectory (Thirdparty/g2o)`
 and
`include_directories (`
	`...`
`# Third parties`
	`Thirdparty`
	`Thirdparty/g2o`
`)`

A quick search online narrowed down the problem to the file core/optimizable_grap.cpp. Indeed, there is a function addVertex defined here, but it has not the same input arguments. These are 
`bool addVertex(HyperGraph::Vertex* v, Data* userData)`. 
I found the file [here](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/optimizable_graph.cpp#L257). You can see that the second argument has not the required type: `Data` instead of `HyperGraph::Data`

Like a pure noob, I added `HyperGraph` in front of `Data` in the `optimizable_file.cpp` and in the corresponding header file but the compilation crashes anyway with that error message: 

`g2o::HyperGraph::Data has not been declared`
 
Does anyone have any idea how I can fix this issue? 

## Specifications

  - Ubuntu version: 16.04 Xenial
  - ROS Version: Kinetic
  - Autoware branch: Master
 
Thank you very much for your help ! 
Guillaume

GuillaumeHauss, 2017-10-30:
Hi all, 

I've done some digging on this error, and here's what I've found so far. 
References to `addVertex` function in file `Optimizer.cc` are 

1. in function `BundleAdjustment`
`l.81` with argument of type `VertexSE3Expmap`
`l.100` with argument of type `VertexSBAPointXYZ`
2. in function `PoseOptimization` 
`l.259` with argument of type `VertexSE3Expmap`
3. in function `LocalBundleAdjustment`
`l.530` and `l.543` with argument of type `VertexSE3Expmap`
`l.580` with argument of type `VertexSBAPointXYZ`
4. in function `OptimizeEssentialGraph`
`l.841` with argument of type `VertexSim3Expmap`
5. in function `OptimizeSim3`
`l.1083` with argument of type `VertexSim3Expmap`
`l.1123` and `l.1131` with argument of type `VertexSBAPointXYZ`

One reference to the function `removeVertex` is also flagged as `undefined` in function `BundleAdjustment (l.178)` with argument of type `VertexSBAPointXYZ`. 

So I checked out these types, as they were not `HyperGraph::Vertex` as described in the functions definitions. Here's what I found:
1. `VertexSBAPointXYZ` extends `public BaseVertex<3, Vector3d>`
2. `VertexSE3Expmap` extends `public BaseVertex<6, SE3Quat>`
3. `VertexSim3Expmap` extends `public BaseVertex<7, Sim3>`

I don't know if I have to keep digging into it, to find the type error somewhere.. 
Any help would be appreciate
Thanks

dejanpan, 2017-10-31:
@sujiwo can you please help here?

sujiwo, 2017-10-31:
Yes @dejan, I'm currently trying to merge new updates in ORB-SLAM, but I
couldn't promise any timeline.

GuillaumeHauss, 2017-10-31:
Thank you very much for helping me out!

hvpandya, 2017-11-30:
Seems like you already have a version of lib2go installed on your system. Try removing the default version:

`apt-get remove ros-kinetic-lib2go`

Then try to build the orb_localizer package

`catkin_make --pkg orb_localizer`

huckl3b3rry87, 2018-03-10:
I am running into a similar issue here https://github.com/CPFL/Autoware/issues/1147

and when I run 
```
$ sudo apt-get remove ros-kinetic-lib2go
```
I get
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-lib2go
```

closed 2018-04-07 by amc-nu