autowarefoundation/autoware #868: euclidean_cluster package with Gazebo

huckl3b3rry87, 2017-11-03:
As I mentioned [here](https://github.com/CPFL/Autoware/issues/758) am trying to use Gazebo and I am running into some issues that may be either due to the fact that I am a new user or because (as it says in the README.md, Gazebo is not supported for ROS-kinetic).  

## Steps to Reproduce the Problem

  1.  Launch Gazebo under the Simulation tab
  2. Click on the euclidian_cluster app 

Then I get this error:
```
[ INFO] [1509676972.763606801, 616.440000000]: vectormap_filtering: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[ INFO] [1509676972.764039357, 616.440000000]: Euclidean Clustering : 0.000000
Could not estimate a planar model for the given dataset.
[ INFO] [1509676972.803600916, 616.480000000]: vectormap_filtering: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[ INFO] [1509676972.803653652, 616.480000000]: Euclidean Clustering : 0.000000
Could not estimate a planar model for the given dataset.
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 0!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 0!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
```
Looking at the ``tf``

[image]

It can be seen that indeed, the ``map`` frame is not connected to the ``base_link``. Should the ``mobility`` link be connected to the ``base_link``? 

While I could go mucking around to try and fix this, I do not want to do something that is silly because I am a new user of Autoware and help would be much appreciated! Thanks!

## Specifications
  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: master
  - Gazebo Version: 8

dejanpan, 2017-11-10:
@huckl3b3rry87 I am afraid that no-one is really using this version of gazebo to be able to help off the bet. I think that it would really make sense that completely switch to Gazebo 8 as mentioned in #758. We could definitely help integrating it with the current state of topics and services in Autoware.

huckl3b3rry87, 2017-11-10:
So, is Gazebo 7 working fine with Autoware? Is this tree connected etc.?

dejanpan, 2017-11-11:
@huckl3b3rry87 the simplest answer is that we do not know as noone had enough time to give it enough test. That is also why I would not bother it and go directly to Gazebo 8 (https://www.osrfoundation.org/simulated-car-demo/) which OSRF supports.

huckl3b3rry87, 2017-11-11:
@dejanpan thank you for your response. I am familiar with car_demo, I actually posted an issue a while ago [here](https://github.com/osrf/car_demo/issues/19). Among other things, If you look at my first post under 1. ,, long story short; I was hoping that we could build car_demo into something exactly like Autoware, but that was before I found Autoware. I really like what you guys (and gals) have done here and think it is better for me to start here.

I was simply wondering if it works with Gazebo 7. Personally, I do not have a strong affinity towards Gazebo 8 and if Gazebo 7 is working fine, I might consider just doing that.

amc-nu, 2017-11-17:
@huckl3b3rry87 the error shown in Euclidean cluster basically means the input cloud is empty (There are no points in the received message)
I haven’t used Gazebo so I really cannot help you there.

closed 2017-11-27 by amc-nu