autowarefoundation/autoware #871: when launch motion plan, report error :enablePlannerDynamicSwitch arg to be set

asimay, 2017-11-06:
when load motion launch as quick_start.pdf instruction, ti prompt error on screen: enablePlannerDynamicSwitch arg to be set.

[image]
I look the code and find it lack of arg definition.
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/motion/packages/astar_planner/launch/velocity_set.launch

should add :  `<arg name="enablePlannerDynamicSwitch" default="true" />`
here. or else no motion plan effect on rviz.

# Bug
## Expected Behavior
motion plan should work.

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

dejanpan, 2017-11-10:
@asimay could you submit a PR? 
@TomohitoAndo it seems that @asimay is right

asimay, 2017-11-13:
@dejanpan,  ok. please see here:

https://github.com/CPFL/Autoware/pull/892

closed 2017-11-21 by yk-fujii