autowarefoundation/autoware #887: [Bug] rebuild some packages that has not been modified

YamatoAndo, 2017-11-10:
# Bug
If the package has not been modified, it should not be rebuilt.

| before | after |
|--------|-------|
| [image] | [image] |

## Steps to Reproduce the Problem

```sh
$ cd <your-Autoware-path>/ros
$ catkin_make # build all packages, if it is first time build
$ catkin_make # rebuild some packages
```
## Specifications

  - Ubuntu version: 16.04
  - ROS Version: Kinekic
  - Autoware branch: develop

aohsato, 2017-11-10:
[image]
After this modify, gitignore for qt generated files is not enabled... Would you please fix?

YamatoAndo, 2017-11-13:
Perhaps, I think those files were created before my commit. Plz delete those files and try building again.

aohsato, 2017-11-13:
@YamatoAndo Okey, ui/moc files are generated in build directory by your modification. THX!
[image]

closed 2017-11-13 by aohsato