autowarefoundation/autoware #898: Visualization sensor_msgs/NavSatFix message in rviz

hakuturu583, 2017-11-14:
# New Feature
add OverlayGpsDiaplay and vislalize sensor_msgs/NavSatFix message in rviz by using Google Static Map API.

dejanpan, 2017-11-25:
@hakuturu583 do you plan to add this feature or would you like that someone from Autoware team adds it? If the latter - can you provide some pointers to the documentation or how to do it?

I will for now close the issue, please re-open if you can provide more details.

hakuturu583, 2017-12-21:
@dejanpan Sorry for my late reply. I choose latter and I will add detailed documentation about my Pull Request. Please wait.
Thank you.

asimay, 2018-01-02:
hi, all, I test this package. It works good! it can show your GPS trajectory on google map, by this tool, you can clearly see your trajectory is accordance with your actual trajectory or not. it's very well.

but only 1 issue, the package works very slow for bag data, when show gps trajectory, it almost occupy all PC resources, this leads to PC is running very slow. I think this should be optimized.

hakuturu583, 2018-01-02:
@asimay I made a Message Per Plot property in this plugin.(see:https://github.com/hakuturu583/gps_rviz_plugin)
This plugins recieve sensor_msgs/NavsatFix message more than onec and after that, call the Google Static Map API.
So if you feel this plugin works slowly, please increase Message Per Plot property.

amc-nu, 2018-04-07:
Please feel free to open this again @hakuturu583 if you plan to add this feature.

closed 2018-04-07 by amc-nu