autowarefoundation/autoware #911: cannot launch node of type [ndt_localizer/ndt_matching_omp]

dlounis, 2017-11-19:
Dear all,

I have installed AW on ubuntu 14.04 and I'm trying to reproduce the quick start example. The display does not follow the car frame. I have found those errors :

On one of the terminals :
ERROR: cannot launch node of type [ndt_localizer/ndt_matching_omp]: can't locate node [ndt_matching_omp] in package [ndt_localizer]

On rVIZ : (under raw_points) :
For frame [velodyne]: No transform to fixed frame [world].  TF error: [Could not find a connection between 'world' and 'velodyne' because they are not part of the same tree.Tf has two or more unconnected trees.]

Thanks a lot for your help.
# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

YamatoAndo, 2017-11-20:
Thank you for the report.
Is "Use OpenMP" turned on in the app tab of ndt_matching?
Please turn it off and try it.
If it is off, ndt_matching is executed instead of ndt_matching_omp.

By the way, when PCL version is less than 1.7.2, ndt_matching_omp is not created.
(You can solve it by raising the PCL version and building Autoware)

dlounis, 2017-11-20:
Thanks for the quick answer.

As far as I understand, the Quick Start example is here to allow a "quick" start, meaning we do not have to change the default settings on the runtime manager. I have followed exactly the steps provided by the Quick Start document. In this document we only need to specify file locations (map, localization, config for rViz). I have not seen anywhere on this document that we had to set options in the "Setting" tab.

Indeed, there is a link with the pcl_version too (I have looked into the CMakeList files). What happens if PCL version is NOT less than 1.7.2 and there is no OMP ?

Regards

YamatoAndo, 2017-11-20:
I checked my_launch.sh,
I understood that it is not enough for the current Autoware version.
(We overlooked because we normally do not debug with Quick Start)

Please fix as follows and try again.

my_launch.sh
`90:  #<arg name = "use_openmp" value = "true" />`
`91:  <arg name = "use_openmp" value = "false" />`

> What happens if PCL version is NOT less than 1.7.2 and there is no OMP?

Although it seems that there is a problem, it checks with #ifdef _ OPENMP on the source code, so if there is not OpenMP, the program which does not use OpenMP is generated.
However, to avoid confusion, it seems better not to generate ndt_matching_omp without OpenMP by checking OPENMP_FOUND with CMakeLists.

I will fix these problems.
Thank you for the report.

dlounis, 2017-11-20:
Thanks a lot.

It finally worked :). 

Now my question regards the green spheres and the path on the road that do not appear on the image (for both ubuntu 14 and ubuntu 16). I have set the mission and motion launch files as described in the QS. How to display these green spheres and also the green area on the pedestrian crossing (as seen on this video : https://www.youtube.com/watch?v=NDNcy0C-Has

Also, for the mission launch, the numbers on the path for me disappear after some seconds (the decrease and disappear when they reach zero) and I do not have the red little cylinders as well.

Finally when I activate the motion launch, I do not see the curves on the path.

Thanks a lot.

YamatoAndo, 2017-11-21:
I confirmed to reproduce it in my environment.
We found a bug and fixed it.
Please git pull and try again.

Thank you for the report.

dlounis, 2017-11-21:
You're welcome.
Any suggestions for the other issues I mentioned in my previous question ?
Thanks.

dlounis, 2017-11-21:
Hello,

for the needs of a meeting with my research team, I'd like to show all the interesting features in Autoware. Is this possible with the quick start example (can I run the computing tools/packages with that example only?) ?

Please, I'd like to bring to your attention that I have a CPU_only machine with no GPU/CUDA and that I did not build AW against SSD, RCNN nor Yolo2 (that's because I was getting build erros with those nodes).

Many thanks.

YamatoAndo, 2017-11-22:
I think that quick start is sufficient to show Autoware to your research team.
Quick start has all the basic functions of self-driving.

If you want to try another algorithm or new function that is in Autoware, you need to select the node from runtime_manager.
(Please look for it because I think that how to use each function is published on youtube)

dlounis, 2017-11-22:
Thanks.

I still need to know why in my case I cannot see the following details : 

* for mission_launch  : Green spheres, Green pedestrian crossing, 

* for motion_launch : curves on the path

When I activate mission and motion launch files on the QS interface, it seems that this has no effect on the displayed image.

Thanks.

YamatoAndo, 2017-11-22:
Thanks and I'm sorry many times.

I fIxed it.
Please download my_launch.sh and try again.
http://db3.ertl.jp/autoware/sample_data/my_launch.sh

dlounis, 2017-11-22:
No problem. I'll try it later and let you know.

Thanks a lot.

dejanpan, 2017-11-25:
@andoh104 is http://db3.ertl.jp/autoware/sample_data/my_launch.sh something that we should merge int develop?

@dlounis can we close this issue? Were you able to get enough help?

dejanpan, 2017-11-25:
@dlounis can you say which research team are you with?

dlounis, 2017-11-25:
not yet. I still have difficulties with many features. Now even the QS demo does not work. I'll provide feedback asap. Thanks.

dlounis, 2017-11-25:
Hello,
could someone help me with this issue. The car loses its track most of the time and I am no longer able to display the digits on the path when activating the mission launch nor I can display other details like green area on the path...
thanks.

MaskedName, 2017-12-05:
I am also a user and I am trying to figure out how to run autoware. From my experience, if the CPU computation power is not enough, the car will lose its track. And many strange things will happen, for example, detected objects are not in the right position, map not displayed, ...

dlounis, 2017-12-05:
You are right. I have tried with a more powerful computer and quick start example works. I see that you managed to detect objects ? Please, do you mean 2D object detection on 2D color images ? As far as I know there is no video data included in the moriyama bag file. I have downloaded Kitti data and managed to run the kitti_player but I do not see any object detection (no bounding boxes displayed).

Thanks for your answer.

MaskedName, 2017-12-05:
I have not run kitti_player yet.

closed 2018-04-07 by amc-nu