autowarefoundation/autoware #914: Mapping with KITTI

aztinki, 2017-11-20:
Was anyone able to perform mapping in Autoware for kitti datasets? If so, how? Because when I try to map, no pointcloud is seen in rviz. 

## Specifications

  - Ubuntu version: 14.04
  - ROS Version: Indigo
  - Autoware branch: master

dejanpan, 2017-11-25:
@aztinki can you be a bit more precise? which dataset are you referring to? images or PC?

We can play the data back: https://github.com/CPFL/Autoware/tree/develop/ros/src/util/packages/kitti_pkg

and we have mapping/localization pkgs:

https://github.com/CPFL/Autoware/tree/develop/ros/src/computing/perception/localization/packages

@amc-nu or @andoh104 might know more.

amc-nu, 2017-11-25:
Using the kitti player mentioned by @dejanpan a map from any of the sets can be generated.
Please refer to page 15 of Tier4 Academy in the Autoware manuals repo.

closed 2017-11-27 by amc-nu

dejanpan, 2017-11-27:
@aztinki here the link: https://github.com/CPFL/Autoware-Manuals/blob/master/en/Autoware_TierIV_Academy_v1.1.pdf

aztinki, 2017-11-27:
Thank you! 
My rosbag was erroneous earlier.

dlounis, 2018-01-09:
Hello,

@aztinki did you manage to use ndt_mapping with kitti bag files?

I tried with a bag file obtained from transforming kitti data to rosbag (https://github.com/tomas789/kitti2bag).

I have followed the procedure described in AW tutorial on generating a map while playing a rosbag. I have renamed the topic kitti/velo/pointcloud to /points_raw but it did not work : I get this error : 

Transform [sender=unknown_publisher]
For frame [velo_link]: No transform to fixed frame [map]. TF error: [Could not find a connection between 'map' and 'velo_link' because they are not part of the same tree.Tf has two or more unconnected trees.]

Regards

kitsukawa, 2018-01-10:
@dlounis I think the frame id of velodyne scan should be "velodyne" instead of velo_link.
Can you create rosbag again after modifying the code below(velo_link -> velodyne)so that the frame id of velodyne scan to be "velodyne".
https://github.com/tomas789/kitti2bag/blob/master/bin/kitti2bag#L317

dlounis, 2018-01-11:
@kitsukawa 
I managed to fix the issue with the map frame... and velo frame name.
Unfortunately I still cannot save map to PCD file... I don't know what I am doing wrong???

kitsukawa, 2018-01-15:
@dlounis Do you use ndt_mapping or approximate_ndt_mapping? approximate_ndt_mapping automatically outputs PCDs in ~/.ros directory every N meters. If you use ndt_mapping, to output pcd, you need to open app, specify the directory and file name and click PCD OUTPUT. See the video below.
https://www.youtube.com/watch?v=ss6Blrz23h8

dlounis, 2018-01-15:
What's the difference between NDT and Approximate_NDT ? Is the map provided in Moriyama data sample (for demos) built using NDT or Approximate_NDT ?

Thanks.

kitsukawa, 2018-01-16:
@dlounis ndt_mapping and approximate_ndt_mapping are the same algorithm for scan matching. As I said in #1028, they are different where to load map as reference. Regarding sample Moriyama data, the pointcloud of moriyama is collected by special measurement system called Mobile Mapping System. It is provided by other survey company. The data is not built using Autoware.

kerimyener, 2018-03-04:
Hi all,
I was wondering if anyone manages to run ndt matching with kitti data. if so, please give me hints. Any help will be highly appreciated.