autowarefoundation/autoware #915: nvcc fatal   : Stray '"' character CMake Error at gpu_euclidean_clustering_generated_gpu_euclidean_clustering.cu.o.Release.cmake:222 (message):   Error generating

jinfagang, 2017-11-21:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug

```
[ 12%] Built target _vector_map_server_generate_messages_check_deps_GetWayArea
CMake Warning (dev) at gpu_euclidean_clustering_generated_gpu_euclidean_clustering.cu.o.Release.cmake:81:
  Syntax Warning in cmake code at column 78

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

Scanning dependencies of target people_msgs_generate_messages_lisp
nvcc fatal   : Stray '"' character
CMake Error at gpu_euclidean_clustering_generated_gpu_euclidean_clustering.cu.o.Release.cmake:222 (message):
  Error generating
  /home/robox/Autoware/ros/build/computing/perception/detection/packages/lidar_tracker/CMakeFiles/gpu_euclidean_clustering.dir/nodes/euclidean_cluster/./gpu_euclidean_clustering_generated_gpu_euclidean_clustering.cu.o

```
this is the second time I occured this problem, is there no one got this??? I never make it success.
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. installed the dependencies
  1. sourced ros
  1. init workspace
  1. catkin_make_release

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: master

andoh104, 2017-11-21:
I can build the current master branch successfully with the following conditions:
- Ubuntu version: 16.04
- ROS Version: kinetic
- Autoware branch: master
- Cuda 9.0

```
[ 12%] Built target _vector_map_server_generate_messages_check_deps_GetUtilityPole
Scanning dependencies of target _vector_map_server_generate_messages_check_deps_GetWayArea
[ 12%] Built target _vector_map_server_generate_messages_check_deps_GetWayArea
Scanning dependencies of target _vector_map_server_generate_messages_check_deps_GetNode
[ 12%] Built target _vector_map_server_generate_messages_check_deps_GetNode
Scanning dependencies of target _vector_map_server_generate_messages_check_deps_PositionState
```

But In the develop branch, there are some problems. #910

jinfagang, 2017-11-22:
I just cloned the codes from master not develop. Is it a CUDA9  problem? I am currently using CUDA8

dejanpan, 2017-11-25:
@andoh104 where are current best instructions for installing CUDA9?

@jinfagang could you try with CUDA9?

andoh104, 2017-11-27:
>where are current best instructions for installing CUDA9?

I think you can install CUDA9 from here:
https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&target_distro=Ubuntu&target_version=1604

But I have an another environment as follows, and I can build the master branch successfully in this environment.
-    Ubuntu version: 14.04
-    ROS Version: Indigo
-    Autoware branch: master
-    Cuda 8.0

```
[100%] Built target lane_rule
Linking CXX executable /home/tandoh/Autoware/ros/devel/lib/lattice_planner/lattice_trajectory_gen
[100%] Built target lattice_trajectory_gen
Linking CXX executable /home/tandoh/Autoware/ros/devel/lib/lattice_planner/lattice_twist_convert
[100%] Built target lattice_twist_convert
~/Autoware/ros$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2016 NVIDIA Corporation
Built on Tue_Jan_10_13:22:03_CST_2017
Cuda compilation tools, release 8.0, V8.0.61
~/Autoware/ros$ 
```

mode89, 2018-01-26:
Hi, everyone!

I'm having the same issue.

I think the problem is in the CMake scripts.
If you look to the line 81 of `gpu_euclidean_clustering_generated_gpu_euclidean_clustering.cu.o.Release.cmake`, you can see something like that:
```
set(CUDA_NVCC_COMPILE_DEFINITIONS " ... ;ROS_PACKAGE_NAME="lidar_tracker"; ... ")
```
As we can see the configuration of quotation marks is wrong. Replacing `"lidar_tracker"` with `'lidar_tracker'` eliminates the error.

I think the possible workaround would be removing of `ROS_PACKAGE_NAME` from the list of compile definitions before calling `cuda_add_library()`.

`rbsspfvehicletracker.cu` produces the same error.

My configuration is:
* Ubuntu 16.04 arm64
* ROS Version: kinetic
* Autoware branch: master
* CUDA 8.0
* CMake 3.9.2

mylxiaoyi, 2018-03-14:
There is a patch: https://gitlab.kitware.com/cmake/cmake/commit/c1f4f13dbfa7caf6bbf4d8b70a7f09f786c7eed6

Refer to https://github.com/NVIDIA/gvdb-voxels/issues/2

amc-nu, 2018-04-07:
Thanks @mylxiaoyi

closed 2018-04-07 by amc-nu

q8wwe, 2020-03-30:
> There is a patch: https://gitlab.kitware.com/cmake/cmake/commit/c1f4f13dbfa7caf6bbf4d8b70a7f09f786c7eed6
> 
> Refer to [NVIDIA/gvdb-voxels#2](https://github.com/NVIDIA/gvdb-voxels/issues/2)

how to use it??