autowarefoundation/autoware #916: Velodyne doesn't display on the 3D map

aalzooke, 2017-11-21:
[Fill-out] only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
The Velodyne should work on the 3D Map 

## Actual Behavior
The Velodyne does't show on the 3D map 

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

yk-fujii, 2017-11-22:
Please tell me more detailed situation.

Is this a ubuntu16.04?master branch?ros indigo?
How did you do it?

aalzooke, 2017-11-22:
Yes it's Ubuntu 16.04 and it's ROS Indigo 
The veldoyne it's working only when i display the 3D map it doesn't show on RVIZ Map

dejanpan, 2017-11-25:
@aalzooke this can not be, ROS indigo is not supported and released on 16.04: http://www.ros.org/reps/rep-0003.html#indigo-igloo-may-2014

In addition please provide exact steps in how are you trying to launch velodyne driver and visualize it (e.g. which launch files do you run).

amc-nu, 2017-11-27:
probably NDT is not working. Check TF Tree using RQT, if the tree is not connected, make sure NDT is receiving data from the Velodyne and the initialization is correct. Check the manual for extra details. https://github.com/CPFL/Autoware-Manuals/tree/master/en

closed 2017-11-27 by amc-nu