autowarefoundation/autoware #918: [Solved] Use the CPFL/Autoware GUI to launch IMU Xsens MTi-G-7XX file

yxh0606, 2017-11-23:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
Use the CPFL/Autoware GUI to launch IMU Xsens MTi-G-7XX file

## Actual Behavior
It has the Xsens MTi-300 driver for Xsens driver.  By the way, it doesn't have the Xsens MTi-G-7XX driver yet.

## Steps to Reproduce the Problem

## Part 1: Use the Ubuntu to launch the MTi-G-7XX driver.

 Step 1: Download Synaptic Package Manager in the Ubuntu Software
 Step 2: Install “ros-kinetic-xsens-driver” in the Synaptic Package Manage
 Step 3: $ lsusb          (check the IMU Xsens MTi-G-7XX USB info)
 Step 4: $ roslaunch xsens_driver xsens_driver.launch 
 Step 5: $ rostopic list   
              It will us the /imu/data in the topic list.

## Part 2: Use the Ubuntu Autoware to launch the  MTi-G-7XX driver.

 Step 6: adding following the part in
~/Autoware/ros/src/ulti/packages/runtime_manager/scripts/sensing.yaml

- name : Xsens MTi-G-7XX
        desc : Xsens MTi-G-7XX with GNSS/INS desc sample
        probe:
        run  : roslaunch xsens_driver xsens_driver.launch
        param: 
        gui  :
          flags: [ kill_children ]
          baud : { flags: [ nl ] }
          mode : { flags: [ nl ] }
          frequency: { flags: [ nl ] }

 Step 7: $ cd Autoware/ros/src/sensing/drivers/imu/packages/xsens/src/
 Step 8: $  rm -r xsens
 Step 9: $ cd Autoware/ros
Step 10:$ ./catkin_make_release
Step 11:$ ./run 
Step 12:$ rostopic list 
It will show us the /imu/data in rostopic.

## Specifications

  - Ubuntu version: Ubuntu Linux 16.04 64bit
  - ROS Version: ROS Kinetic
  - Autoware branch:  

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

dejanpan, 2017-11-25:
@kitsukawa do you understand what @yxh0606 is trying to tell? Trying to launch a different type of an IMU?

yxh0606, 2017-11-26:
@dejanpan Many thanks for the prompt reply. Here is the temporary solution to launch the IMU Xsens MTi-G-7XX.   Hope the contributor could add the MTi-G-7XX in the source code consequently.

kitsukawa, 2017-11-29:
@yxh0606 Does Xsens driver in Autoware work for MTi-G-7xx? Can you check the operation of the Xsens driver in Autoware/ros/src/sensing/drivers/imu/xsens using your sensor(MTi-G-7xx)? If it does not work, we can use the driver in the Synaptic package manager instead and can add the checkbox for launching the driver.

yxh0606, 2017-11-29:
@kitsukawa Thank you for your replying. 
No, it is not compatible with Xsens MTi-G-7xx driver from your original code.

But your original source code,  the Autoware work well for Xsens MTi-300 driver.  
It will run the Autoware/ros/src/util/packages/runtime_manager/scripts/mti300.sh file.

My initial idea is same with you: just use the driver in Synaptic package manager, it works well with Xsens MTi-G-7xx driver.

Then I modify the code, it works well with checkbox for launching the driver in Autoware/run_time manager GUI.

kitsukawa, 2017-11-29:
@yxh0606 That's good. If it is possible, can you make a PR of your modification so that others can use Xsens MTi-G-7xx with Autoware? It would be great if you could include the installation guide of Xsens MTi-G-7xx driver in Synaptic package manager.

yxh0606, 2017-11-29:
@kitsukawa Understand.  The details to use the Xsens MTi-G-7xx with Autoware is under "issue/[Solved] Use the CPFL/Autoware GUI to launch IMU Xsens MTi-G-7XX driver #918". 

Please follow the steps posted in the #918 to use the Xsen MTi-G-7xx driver.
Part 1: Use the Ubuntu to launch the MTi-G-7XX driver.
Part 2: Use the Ubuntu Autoware to launch the MTi-G-7XX driver.

Thank you for your kind help.

closed 2018-02-20 by kitsukawa