autowarefoundation/autoware #919: Runtime manager does not display when ./run is tried

kimjongyoung, 2017-11-24:
It did display but now it doesn't. Here's what I see on the terminal (Ubuntu 16.04)

... logging to /home/bayesian/.ros/log/7fd194ee-d0b7-11e7-ad5d-4439c4519f9c/roslaunch-bayesian-3710.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bayesian:37281/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [3741]
ROS_MASTER_URI=http://bayesian:11311/

setting /run_id to 7fd194ee-d0b7-11e7-ad5d-4439c4519f9c
process[rosout-1]: started with pid [3754]
started core service [/rosout]

yk-fujii, 2017-11-24:
Could you send me a log when the following command is executed?

`
source devel/setup.bash; roscore & rosrun runtime_manager runtime_manager_dialog.py
`
if you not using bash, please select other setup file.(setup.sh, setup.zsh)

kimjongyoung, 2017-11-24:
Thank you so much for your help;

here is the output

bayesian@bayesian:~/Downloads/Autoware/ros$ source devel/setup.bash;
roscore & rosrun runtime_manager runtime_manager_dialog.py
[1] 5430
... logging to /home/bayesian/.ros/log/a19a23a6-d0bd-11e7-ad5d-4439c4519f9c
/roslaunch-bayesian-5430.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/home/bayesian/Downloads/Autoware/ros/src/util/packages/
runtime_manager/scripts/runtime_manager_dialog.py", line 33, in <module>
    import wx
ImportError: No module named wx
bayesian@bayesian:~/Downloads/Autoware/ros$ started roslaunch server
http://bayesian:44721/
ros_comm version 1.12.7

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES

auto-starting new master
process[master]: started with pid [5449]
ROS_MASTER_URI=http://bayesian:11311/

setting /run_id to a19a23a6-d0bd-11e7-ad5d-4439c4519f9c
process[rosout-1]: started with pid [5462]
started core service [/rosout]

----------------------------
Looking at the error message; ImportError: No module named wx,
I checked

bayesian@bayesian:~$ apt list --installed | grep python-wxtools

WARNING: apt does not have a stable CLI interface. Use with caution in
scripts.

python-wxtools/xenial,xenial,now 3.0.2.0+dfsg-1build1 all [installed]

yk-fujii, 2017-11-24:
it seems that you not installed wx python module.
Please install wx with following command.

`
sudo apt install python-wxgtk 3.0
`

kimjongyoung, 2017-11-24:
Dear Mr. Fujii,
it says python-wxgtk3.0 is already installed.

yk-fujii, 2017-11-24:
ummh....
What version are you using python?
Is `import wx` valid in your python?

kimjongyoung, 2017-11-24:
No, it says "no module named wx"
Is it a sudo issue?

yk-fujii, 2017-11-29:
it means your environment have not installed wxpython corresponding to python you're using.
I think that it is a problem of python version.

What is the version of python used?
Could you reinstalled wxpython?

closed 2017-12-11 by amc-nu

datnguyen82, 2019-05-30:
#736