autowarefoundation/autoware #933: lane select underlying bug.

asimay, 2017-11-28:
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/mission/packages/lane_planner/nodes/lane_select/lane_select_core.cpp#L632

at line 627, here defined `current_lane_idx_ = -1;` , so it is always execute line 632,  always initialize lane select, never touch `processing()`.
please check.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

asimay, 2017-11-28:
should remove line 627-630.

asimay, 2017-11-29:
any update? I can submit a PR for this bug.

yk-fujii, 2017-11-29:
thanks for report!
It certainly looks like it will not be right.

@TomohitoAndo 
could you check it please?

asimay, 2017-12-05:
hi, dear, it seems no response of this issue...

asimay, 2017-12-09:
hi dear, any update? I think there is need someone to maintain the debug code. 
if ok, i'm also glad to join in your team and maintain.

shinpei0208, 2017-12-10:
That would be appreciated if you can contribute to the repo.

@TomohitoAndo @aohsato Can you check the bug?

dejanpan, 2017-12-11:
@asimay thx for the report. 

@shinpei0208 @TomohitoAndo it seems that with this bug we will never be able to do lane change: https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/mission/packages/lane_planner/nodes/lane_select/lane_select_core.cpp#L170, correct?

closed 2020-05-29 by gbiggs