autowarefoundation/autoware #934: Build hexacam_node make Error

arashroomi, 2017-11-29:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

Hi I am doing camke of the Autoware main branch,it gives me Built error as below. Apparently it has issues on making the Hexcam package. I ran catkin_make --pkg hexacam command separately to see if it can build it separately. It was able to be built but again in the ./catkin_make_release it gives the same error. What is wrong?

[  1%] Built target op_utility
ptxas warning : Stack size for entry function 'dt1d_x' cannot be statically determined
ptxas warning : Stack size for entry function 'dt1d_y' cannot be statically determined
ptxas info    : 22 bytes gmem
ptxas info    : Compiling entry function 'calc_norm' for 'sm_35'
ptxas info    : Function properties for calc_norm
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 32 registers, 356 bytes cmem[0], 2 textures
ptxas info    : Compiling entry function 'dt1d_x' for 'sm_35'
ptxas info    : Function properties for dt1d_x
    8 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 32 registers, 432 bytes cmem[0]
ptxas info    : Function properties for _Z9dt_helperPfS_Piiiiiiff
    56 bytes stack frame, 52 bytes spill stores, 52 bytes spill loads
ptxas info    : Compiling entry function 'calc_a_score' for 'sm_35'
ptxas info    : Function properties for calc_a_score
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 22 registers, 400 bytes cmem[0], 4 bytes cmem[2]
ptxas info    : Compiling entry function 'process_part' for 'sm_35'
ptxas info    : Function properties for process_part
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 27 registers, 388 bytes cmem[0], 2 textures
ptxas info    : Compiling entry function 'resize' for 'sm_35'
ptxas info    : Function properties for resize
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 11 registers, 364 bytes cmem[0], 2 textures
ptxas info    : Compiling entry function 'calc_hist' for 'sm_35'
ptxas info    : Function properties for calc_hist
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 27 registers, 360 bytes cmem[0], 44 bytes cmem[2], 4 textures
ptxas info    : Compiling entry function 'inverse_Q' for 'sm_35'
ptxas info    : Function properties for inverse_Q
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 21 registers, 392 bytes cmem[0]
ptxas info    : Compiling entry function 'dt1d_y' for 'sm_35'
ptxas info    : Function properties for dt1d_y
    8 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 32 registers, 428 bytes cmem[0]
ptxas info    : Function properties for _Z9dt_helperPfS_Piiiiiiff
    56 bytes stack frame, 52 bytes spill stores, 52 bytes spill loads
ptxas info    : Compiling entry function 'process_root' for 'sm_35'
ptxas info    : Function properties for process_root
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 27 registers, 388 bytes cmem[0], 2 textures
ptxas info    : Compiling entry function 'calc_feat' for 'sm_35'
ptxas info    : Function properties for calc_feat
    320 bytes stack frame, 320 bytes spill stores, 320 bytes spill loads
ptxas info    : Used 32 registers, 376 bytes cmem[0], 40 bytes cmem[2], 3 textures
[  1%] Built target cuda_binary
In file included from /home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:46:0:
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp: In member function ‘void HexaCam::parse_packet()’:
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:498:52: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  uint32_t packet_id   = ntohl(*(uint32_t*)(packet+0));
                                                    ^
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:570:20: warning: unused variable ‘frame’ [-Wunused-variable]
     unsigned char *frame = (*curmsg)[i].data.data();
                    ^
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:498:11: warning: unused variable ‘packet_id’ [-Wunused-variable]
  uint32_t packet_id   = ntohl(*(uint32_t*)(packet+0));
           ^
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:491:11: warning: variable ‘len’ set but not used [-Wunused-but-set-variable]
  unsigned len, addrlen;
           ^
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp: In function ‘int main(int, char**)’:
/home/arash/Autoware/ros/src/sensing/drivers/camera/packages/hexacam/nodes/hexacam/hexacam.cpp:645:6: warning: unused variable ‘count’ [-Wunused-variable]
  int count = 0;
      ^
Linking CXX executable /home/arash/Autoware/ros/devel/lib/hexacam/hexacam_node
[  1%] Built target hexacam_node
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

## Specifications

  - Ubuntu version: 14.04
  - ROS Version: 
  - Autoware branch: Main

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

amc-nu, 2017-11-30:
According to your output, the error is somewhere else. You’re looking just that the hexacam node was the last one to be compiled and linked correctly. 
Look closely in your output to find the actual error. Or share the complete error log.

arashroomi, 2017-11-30:
Is this helpful? I made an error log of the catkin_make. Are there packages I still need to install? The strange thing is that I could install Autoware once and at some point, it stopped working and now I cannot install it anymore.
 
arash@arashtue:~/Autoware/ros$ catkin_make ... >~/Autoware/error.log
fast_pcl requires PCL 1.7.2 or higher versions
fast_pcl requires PCL 1.7.2 or higher versions
'canlib' is not installed. 'can_listener' is not built.
'ENet/Caffe' is not installed. 'image_segmenter_enet' will not be built.
'SSD/Caffe' is not installed. 'ssd_unc' will not be built.
'Darknet' was not found. 'yolo2_wa' will not be built.
CMake Warning at sensing/drivers/camera/packages/pointgrey/CMakeLists.txt:13 (find_package):
  By not providing "FindXercesC.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "XercesC", but
  CMake did not find one.

  Could not find a package configuration file provided by "XercesC" with any
  of the following names:

    XercesCConfig.cmake
    xercesc-config.cmake

  Add the installation prefix of "XercesC" to CMAKE_PREFIX_PATH or set
  "XercesC_DIR" to a directory containing one of the above files.  If
  "XercesC" provides a separate development package or SDK, be sure it has
  been installed.

'flycapture2' is not installed. 'grasshopper3' will not be built.
'lady_bug' is not installed. 'ladybug' will not be built.
-- USING ACCELERATED CLUSTERING --
Version: 7.5
Library: /usr/lib/x86_64-linux-gnu/libcuda.so
Runtime: /usr/local/cuda/lib64/libcudart.so
'BGGAPI' is not installed. 'vlg22c_cam' will not be built.
'SSD/Caffe' is not installed. 'region_tlr_ssd' will not be built.
make: *** No rule to make target `...'.  Stop.
Invoking "make ... -j8 -l8" failed

amc-nu, 2017-11-30:
Not yet. This is not the complete log. These are only notices.

closed 2017-12-06 by amc-nu