autowarefoundation/autoware #937: Failed to play KITTI data

dlounis, 2017-11-30:
Dear all,

I have followed instructions here :
https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/kitti_pkg/README.md

Unfortunately, I did not manage to run the Launch file.

I get those errors :

[ INFO] [1512035912.721070639]: No points node received, defaulting to points_raw, you can use _points_node:=YOUR_TOPIC
[ INFO] [1512035912.722012478]: No camera info topic received, defaulting to /camera/camera_info, you can use _camera_info_topic:=YOUR_TOPIC
[ INFO] [1512035912.723286716]: No projection matrix topic received, defaulting to /projection_matrix, you can use _projection_matrix_topic:=YOUR_TOPIC
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[ERROR] [1512035912.944728516]: Error reading CAMERA02/CAMERA03 calibration
Traceback (most recent call last):
  File "/home/me/Autoware/ros/src/util/packages/kitti_pkg/kitti_box_publisher/scripts/kitti_box_publisher.py", line 21, in <module>
    from cv_tracker.msg import image_obj
ImportError: No module named cv_tracker.msg
[ERROR] [1512035912.954633798]: Error reading CAMERA00/CAMERA01 calibration
[kitti_box_publisher-6] process has died [pid 13031, exit code 1, cmd /home/me/Autoware/ros/src/util/packages/kitti_pkg/kitti_box_publisher/scripts/kitti_box_publisher.py /home/me/Autoware/ros/src/util/packages/kitti_pkg/kitti_player/dataset/2011_09_26/2011_09_26_drive_0093_sync//tracklet_labels.xml __name:=kitti_box_publisher __log:=/home/me/.ros/log/8d5d485e-d5b4-11e7-a040-b8ca3abd3db1/kitti_box_publisher-6.log].
log file: /home/me/.ros/log/8d5d485e-d5b4-11e7-a040-b8ca3abd3db1/kitti_box_publisher-6*.log

I am new to ROS. Maybe I did not source the right setup.bash file ?

Thank you.

## Specifications

  - Ubuntu version: 14.04
  - ROS Version:Indigo

amc-nu, 2017-11-30:
If you wish to see the detection results (bounding boxes) then please pull the develop branch instead. This error was fixed on #865 and is going to be integrated in the coming 1.6 version.
If you don’t need the detection results then you can stay in the master branch and  see the data published in the topics indicated in the README.
If I remember correctly you wish to use SSD/Yolo2. If I’m correct then you can keep this branch and ignore these errors safely.

dlounis, 2017-11-30:
Thanks. I manage to display point clouds and camera data from kitti data. 
I did not know that the kitti launch file performs detection as well. 

Actually I was planning to use the command line to run detection tools (rosrun nodes) and display the output of detection on rviz. Is it possible to perform detection without pulling the develop branch (I am afraid this will alter my current install and I need it to be working for a meeting on monday).

If possible, then I'll be grateful if you can tell me which node to run in order to perform detection and get bounding boxes displayed ?

Regarding SSD and Yolo2, I actually did not build them. The reason is that last time I did that I couldn't build Autoware (it was for kinetic + ubuntu 16. Now I'm on 14.04 + Indigo). Are SSD and Yolo2 required for detection ?

Thank you.

amc-nu, 2017-12-01:
@dlounis Please refer to the detection section (page 32) of the TierIV Academy documentation
https://github.com/CPFL/Autoware-Manuals/blob/master/en/Autoware_TierIV_Academy_v1.1.pdf

In order to build SSD and YOLO2 you need to correctly setup NVIDIA Drivers, CUDA and Caffe (for SSD). Please check the corresponding README https://github.com/CPFL/Autoware#how-to-build

shuzhenzhang, 2017-12-01:
hi,
1.git checkout develop
2../catkin_make_release
3.roslaunch kitti_launch kittiplayer.launch --screen

dlounis, 2017-12-01:
@all 
is it possible to run the former launch file on a terminal and run the nodes that does car and pedestrian detection on other terminals ?

If yes, which are those nodes ?

Thanks a lot.

dlounis, 2017-12-01:
I checked out to develop
no bounding boxes displayed on image/camera topics in rviz.

I do not have ssd/fastrcnn/yolo2 installed. Is this the reason I don't see detection results (bbx's) ?

Which algorithm/node is used to produce those results ?

Thanks.

dlounis, 2017-12-01:
I wonder how I can display those objects (image_obj messages) : 
header: 
  seq: 301
  stamp: 
    secs: 1317043155
    nsecs:  24432076
  frame_id: camera
type: car
obj: 
  - 
    x: 724
    y: 167
    height: 139
    width: 278
    score: 0.648940086365
  - 
    x: 608
    y: 162
    height: 51
    width: 68
    score: -0.368480205536
  - 
    x: 208
    y: 69
    height: 69
    width: 139
    score: -0.460635185242

on the camera image from kitti dataset.

Sindhura-b, 2018-01-11:
@dlounis I got similar output on my terminal, can you explain me what you did next to visualize point clouds from kitti?

closed 2018-02-14 by dlounis