autowarefoundation/autoware #944: errors making develop branch

dlounis, 2017-12-01:
hello,

I checked out to develop branch today in order to see object detection bounding boxes. I have ubuntu 14.04 + indigo. I have also installed cuda 6.5 and ssd. I get these errors when I build autoware : 

/Autoware/ros/src/computing/perception/detection/packages/lidar_tracker/nodes/vscan_lidar_track/mainwindow.cpp:2:27: fatal error: ui_mainwindow.h: No such file or directory
 #include "ui_mainwindow.h"

make[2]: *** [computing/perception/detection/packages/lidar_tracker/CMakeFiles/vscan_lidar_track.dir/nodes/vscan_lidar_track/mainwindow.cpp.o] Error 1

make[1]: *** [computing/perception/detection/packages/lidar_tracker/CMakeFiles/vscan_lidar_track.dir/all] Error 2

make[1]: *** Waiting for unfinished jobs....

Any help ?

Thanks.

andoh104, 2017-12-03:
Did you install Qt5?

dlounis, 2017-12-04:
Hi,

Yes, I have Qt5...

dlounis, 2017-12-04:
I have ubuntu 14.04 and indigo... maybe this is the reason the Develop branch does not build on my system ?

Is the develop branch only compatible with ubuntu 16.04 and Kinetic ???

Thanks.

dlounis, 2017-12-04:
I have both Qt4 and Qt5 on my computer (Ubuntu14.04).
I believe that Qt5 was installed with Autoware. However, I do not remember why Qt4 was installed.

When I enter : pkg-config --modversion QtCore
I obtain 4.8.6.

I don't think this is the reason behind my issue in not being able to build the develop branch.

Indeed I have another computer with the same configuration (Ubuntu 14.04 + indigo). When I enter : 
pkg-config --modversion QtCore
I also obtain the version 4.8.6.... And Autoware works fine (except the object detection bounding boxes that I do not see... the reason why I am here).

Regards.

andoh104, 2017-12-10:
Is this same issue as #947 ?
If so , you'd better close this.

closed 2017-12-10 by dlounis