autowarefoundation/autoware #947: qt4 qt5 conflict : develop branch

dlounis, 2017-12-05:
Dear all,

since I git checked out the develop branch, I get a build error saying : 
 ui_mainwindow.h no such file or directory

I have search the web for solutions and I still can't fix the issue. Any help please ?

Thanks.

andoh104, 2017-12-09:
Can you build the master branch ? 
And about the develop branch, I can build only on CUDA8 environment.  (see #910) 

I guess your Qt environment is something wrong.

This is mine.
```
$ dpkg -l | grep qt5
ii  appmenu-qt5                                              0.3.0+16.04.20170216-0ubuntu1                         amd64        application menu for Qt5
ii  fcitx-frontend-qt5:amd64                                 1.0.5-1                                               amd64        Free Chinese Input Toy of X - Qt5 IM Module frontend
ii  fcitx-libs-qt5                                           1.0.5-1                                               all          transitional dummy package
ii  libaccounts-qt5-1:amd64                                  1.14+16.04.20151106.1-0ubuntu1                        amd64        QT library for single sign on
ii  libdbusmenu-qt5:amd64                                    0.9.3+16.04.20160218-0ubuntu1                         amd64        Qt5 implementation of the DBusMenu protocol
ii  libfcitx-qt5-1:amd64                                     1.0.5-1                                               amd64        Free Chinese Input Toy of X - D-Bus client libraries for Qt5
ii  libqt5clucene5:amd64                                     5.5.1-3build1                                         amd64        Qt 5 CLucene module
ii  libqt5concurrent5:amd64                                  5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 concurrent module
ii  libqt5core5a:amd64                                       5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 core module
ii  libqt5dbus5:amd64                                        5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 D-Bus module
ii  libqt5designer5:amd64                                    5.5.1-3build1                                         amd64        Qt 5 designer module
ii  libqt5designercomponents5:amd64                          5.5.1-3build1                                         amd64        Qt 5 Designer components module
ii  libqt5feedback5:amd64                                    5.0~git20130529-0ubuntu13                             amd64        Qt Feedback module
ii  libqt5gui5:amd64                                         5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 GUI module
ii  libqt5help5:amd64                                        5.5.1-3build1                                         amd64        Qt 5 help module
ii  libqt5multimedia5:amd64                                  5.5.1-4ubuntu2                                        amd64        Qt 5 Multimedia module
ii  libqt5multimediaquick-p5:amd64                           5.5.1-4ubuntu2                                        amd64        Qt 5 Multimedia Quick module
ii  libqt5multimediawidgets5:amd64                           5.5.1-4ubuntu2                                        amd64        Qt 5 Multimedia Widgets module
ii  libqt5network5:amd64                                     5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 network module
ii  libqt5opengl5:amd64                                      5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 OpenGL module
ii  libqt5opengl5-dev:amd64                                  5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 OpenGL library development files
ii  libqt5organizer5:amd64                                   5.0~git20140515~29475884-0ubuntu20                    amd64        Qt PIM module, Organizer library
ii  libqt5positioning5:amd64                                 5.5.1-3ubuntu1                                        amd64        Qt Positioning module
ii  libqt5printsupport5:amd64                                5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 print support module
ii  libqt5qml5:amd64                                         5.5.1-2ubuntu6                                        amd64        Qt 5 QML module
ii  libqt5quick5:amd64                                       5.5.1-2ubuntu6                                        amd64        Qt 5 Quick library
ii  libqt5quickparticles5:amd64                              5.5.1-2ubuntu6                                        amd64        Qt 5 Quick particles module
ii  libqt5quicktest5:amd64                                   5.5.1-2ubuntu6                                        amd64        Qt 5 Quick Test library
ii  libqt5quickwidgets5:amd64                                5.5.1-2ubuntu6                                        amd64        Qt 5 Quick Widgets library
ii  libqt5script5:amd64                                      5.5.1+dfsg-2build1                                    amd64        Qt 5 script module
ii  libqt5scripttools5:amd64                                 5.5.1+dfsg-2build1                                    amd64        Qt 5 script tools module
ii  libqt5sql5:amd64                                         5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 SQL module
ii  libqt5sql5-sqlite:amd64                                  5.5.1+dfsg-16ubuntu7.5                                amd64        Qt 5 SQLite 3 database driver
ii  libqt5svg5:amd64                                         5.5.1-2build1                                         amd64        Qt 5 SVG module
[... cut ...]
ii  libqt4-xmlpatterns:amd64                                 4:4.8.7+dfsg-5ubuntu2                                 amd64        Qt 4 XML patterns module
ii  python-qt4                                               4.11.4+dfsg-1build4                                   amd64        Python bindings for Qt4
ii  qt4-linguist-tools                                       4:4.8.7+dfsg-5ubuntu2                                 amd64        Qt 4 Linguist tools
ii  qt4-qmake                                                4:4.8.7+dfsg-5ubuntu2                                 amd64        Qt 4 qmake Makefile generator tool
$ 
```

yk-fujii, 2017-12-11:
This problem is caused by the version of cmake.
CMAKE_AUTOMOC is needded higher than 3.0.2

@YamatoAndo is fixing it now.

dlounis, 2017-12-11:
Thanks.

Where shall I use qt5_wrap_ui ? in the CMAKEList File ??

Thanks.

amc-nu, 2018-04-07:
you don't need to use it it is already in the CMake file https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/packages/lidar_tracker/CMakeLists.txt#L205

closed 2018-04-07 by amc-nu