autowarefoundation/autoware #948: issue with qt and cuda - lidar_tracker

dlounis, 2017-12-06:
Dear all,

I recently decided to run KITTI_Player in order to apply object detection to 2d images (since there is no video data in Moriyama bag file).

In order to take advantage of my GPU, I also installed CUDA.

Now I have a build error :

/Autoware/ros/src/computing/perception/detection/packages/lidar_tracker/nodes/vscan_lidar_track/mainwindow.cpp:2:27: fatal error: ui_mainwindow.h: No such file or directory
#include "ui_mainwindow.h"

make[2]: *** [computing/perception/detection/packages/lidar_tracker/CMakeFiles/vscan_lidar_track.dir/nodes/vscan_lidar_track/mainwindow.cpp.o] Error 1

make[1]: *** [computing/perception/detection/packages/lidar_tracker/CMakeFiles/vscan_lidar_track.dir/all] Error 2

I have tried to fix this and the only woraround that I've found was to modify lidar_tracker CMAKE file line 135 :

if(EXISTS "usr/local/cuda")

to

if(EXISTS "usr/local/_cuda")

so that it does not enter the if statement.

I believe the problem comes from the lines in the if statement that use Qt (under #QT Stuff comment).

Thanks for your help.

dejanpan, 2017-12-11:
@dlounis we will in couple of weeks make stop to build failures (by enforcing build upon merges). Could you until the report:
- which ubuntu
- which ros
- which autoware branch

dlounis, 2017-12-11:
I have : 
- Ubuntu 14.04
- Ros Indigo
- Develop branch

amc-nu, 2018-04-07:
There is not enough information, but what I can think is that qmake is not correctly installed, be sure you have installed qt 5 (Please check the main README of the repo). or that the version being called in your terminal is 5.

closed 2018-04-07 by amc-nu