autowarefoundation/autoware #957: build error after installing ssdcaffe + cuda

dlounis, 2017-12-07:
# Bug
Hi,

I have built SSD_CAFFE with cuda and when I tried to build Autoware, I get these errors :

make[2]: *** [computing/perception/detection/packages/cv_tracker/CMakeFiles/ssd_unc.dir/nodes/ssd/ssd_node.cpp.o] Error 1
make[1]: *** [computing/perception/detection/packages/cv_tracker/CMakeFiles/ssd_unc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 86%] Built target obj_reproj

Any idea of why does this occur ?

Thank you.
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version: 14.04
  - ROS Version: indigo
  - Autoware branch: develop

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

dejanpan, 2017-12-11:
@dlounis did you use this setup to install:

# NVIDIA Driver Setup
1. Download the RUN file from NVidia's website
https://www.geforce.com/drivers
You'll have a file named similarly to `NVIDIA-Linux-x86_64-XXX.YY.run`

2. Assign execution permissions `chmod +x NVIDIA-Linux-x86_64-XXX.YY.run`

3. Move to a virtual console pressing `Ctrl+Alt+F1` and login

4. Terminate the X Server executing `sudo service lightdm stop`

5. Run the installer `sudo ./NVIDIA-Linux-x86_64-XXX.YY.run`
(**If you are running on a laptop** run instead `sudo ./NVIDIA-Linux-x86_64-XXX.YY.run --no-opengl-files`)

6. Follow the instruction from the wizard. At the end, do not allow the wizard to modify the X configuration.

7. Once back in the console, execute `sudo service lightdm start`. The GUI should be displayed. Login.

8. To confirm everything is set run in a terminal `nvidia-smi`

# CUDA Setup (version 8.0)
1. Download the CUDA Installation **RUN File** from https://developer.nvidia.com/cuda-downloads
You'll have a file named similarly to `cuda_X.0.YY.Z_linux.run`

2. Assign execution permissions `chmod +x cuda_X.0.YY.Z_linux.run`

3. Run the installer `sudo ./cuda_X.0.YY.Z_linux.run`

4. Follow the instructions on screen. DO NOT install the NVIDIA Driver included. Install the CUDA Samples on your home directory.

5. Once finished, to confirm everything is ok. Go to your home directory and execute `cd NVIDIA_CUDA-X.Y_Samples/1_Utilities/deviceQuery` Match X, Y to your CUDA version. i.e. CUDA 9.0 `cd NVIDIA_CUDA-9.0_Samples/1_Utilities/deviceQuery`

6. Compile the sample running `make`

7. Run the sample `./deviceQuery` you should see the details about your GPU(s) and CUDA Setup.

# Caffe Prerequisites
1. `sudo apt-get install libprotobuf-dev libleveldb-dev libsnappy-dev libopencv-dev libhdf5-serial-dev protobuf-compiler`
2. `sudo apt-get install libgflags-dev libgoogle-glog-dev liblmdb-dev`
3. `sudo apt-get install libatlas-base-dev `

# Clone SSD fork of Caffe
1. Go to your home directory
2. Clone the code: `git clone https://github.com/weiliu89/caffe.git ssdcaffe`
3. Move inside the directory `cd ssdcaffe`
4. Checkout compatible API version `git checkout 5365d0dccacd18e65f10e840eab28eb65ce0cda7`
5. Create config file `cp Makefile.config.example Makefile.config`
6. Start building `make`
7. Once completed execute `make distribute`
8. Compile Autoware, Cmake will detect SSD Caffe and compile the SSD nodes.

dejanpan, 2017-12-11:
Note again Cuda version 8.0!

amc-nu, 2018-03-06:
Did you fix this @dlounis ? I’ll close this issue. If you need extra help please feel free to reopen this.

closed 2018-03-06 by amc-nu