autowarefoundation/autoware #958: ssd_unc : error

dlounis, 2017-12-08:
Hi,

I tried to detect cars/objects using kitti data. When I select ssd_unc on AW GUI, I get this error on the terminal :

F1208 09:19:04.427994 23857 io.cpp:45] Check failed: fd != -1 (-1 vs. -1) File not found: data/coco/labelmap_coco.prototxt
*** Check failure stack trace: ***
[ssd-1] process has died [pid 23857, exit code -6, cmd /home/lounis/Autoware/ros/devel/lib/cv_tracker/ssd_unc __name:=ssd __log:=/home/lounis/.ros/log/b7c5c7b2-dbed-11e7-88ae-91ba2f73eaf4/ssd-1.log].
log file: /home/lounis/.ros/log/b7c5c7b2-dbed-11e7-88ae-91ba2f73eaf4/ssd-1*.log
process[ssd-1]: started with pid [2

Any idea of how to fix this ?

It looks like the function io.cpp receives relative incomplete path to labelmap_coco.prototxt from another function which I don't know.

Regards.

MadCauchy, 2018-01-09:
Hi,
i have a similar problem with model VOC0712. Have you solved the problem? Thanks a lot!

amc-nu, 2018-01-27:
Coco models are not currently supported by the node, only the Pascal VOC version indicated in the README file.

closed 2018-01-27 by amc-nu