autowarefoundation/autoware #976: Build error on Ubuntu 16.04 with OpenCV 3.3

daifuku07, 2017-12-13:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
Be able to build correctly if you are using OpenCV 3.x on Ubuntu 16.04.

## Actual Behavior
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/packages/lidar_tracker/CMakeLists.txt#L38
require module "opencv-3.2.0-dev"
So, don't work on different OpenCV 3.x version.

## Steps to Reproduce the Problem

  1. Using OpenCV 3.x (without 3.2.0)
  1. Autoware/ros/catkin_make_release

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: master

# New Feature
Confirm that it works on other than OpenCV 3.2.0, change CMakeLists.txt

130s, 2017-12-13:
Not entirely sure but likely [this regression](https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384/1)?

yk-fujii, 2017-12-13:
I think that it is not a bug.
Autoware refers to OpenCV installed by ROS by default.

daifuku07, 2017-12-13:
I agree that it is not critical critical bug.
But, it is possible that the user has installed the OpenCV version that he uses.
If it works on other than opnecv 3.2.0, I think CmakeLists.txt should be edited.
In my environment(OpenCV 3.3.0), I remove "opencv-3.2.0-dev" on CMakeLists.txt, so build error is not occurred.

sujiwo, 2017-12-13:
You can use installed OpenCV outside ROS. Just set environment variable
CMAKE_PREFIX_PATH before calling catkin or cmake.

yk-fujii, 2017-12-17:
The default version of ROS seems to be changed to 3.3.
We should fix this. Im sorry that I did not grasp it.
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=opencv3

closed 2017-12-18 by daifuku07

wsung1, 2018-01-24:
Hi, @daifuku07 @yk-fujii,
I'm a newbie in here, so please bear with my poor question. While building Autoware, I've run across an error indicating that opencv-3.2.0 is required.
[image]
Now I'm running on
 - ROS 16.04 (kinetic)
 - Autoware 1.6.0
 - OpenCV 2.4.11
In order to solve this, do I have to upgrade OpenCV version to 3.2? How about 3.3?
Any comments will be greatly appreciated.