autowarefoundation/autoware #983: gnss_callback logic is strange when first initial set.

asimay, 2017-12-18:
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp#L532

# Bug
## Expected Behavior
`previous_gnss_pose ` is static variable. 
so in line 544, if first variable at here, the `previous_gnss_pose` is 0 actually,  so at line 560, `offset_x ` variable is meaningless at all.
the logic is strange at first initial step. but after 2nd or later steps, the logic is normal. because the `previous_gnss_pose ` variable is set to meaningful value.

another issue:
and at line 565,  when condition `fitness_score >= 500.0` is true,   `init_pos_set = 1`  will be set many times.

## Actual Behavior
first and 2nd,later handle is meaningful.

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

YamatoAndo, 2017-12-18:
Thanks for your report.
I will fix this bug.

asimay, 2017-12-18:
welcome. glad to contribute to your great package!

closed 2018-02-20 by kitsukawa