autowarefoundation/autoware #986: initialpose_callback() current_pose.z value maybe not be correct.

asimay, 2017-12-18:
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp#L576

# Bug
## Expected Behavior
at line 615, because only compared x & y, not include z,  when run at line 619 or 622, the z will not be the most suitable z if points have similar (x and y) but different z value, the for loop will terminate and won't check z value.

## Actual Behavior
should find the most nearest point of (x,y,z) value.

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

YamatoAndo, 2017-12-18:
Thank you for the report.

I think that you are right.
However, initialpose topic generated from 2D Pose Estimate of RViz does not include z.
Therefore, it is necessary to estimate from x and y.

How do you think that it is better to find the most suitable z?

asimay, 2017-12-18:
dear, @YamatoAndo,  I think you are right.
maybe we can set the most suitable z value to vehicle base_link z value or lidar z value, this is better than very big z value.

YamatoAndo, 2017-12-19:
@asimay To get vehicle base_link z value or lidar z value, you need the "map to base_link" TF. Therefore, it can not be given as the first initial position.

kitsukawa, 2017-12-19:
@asimay Thank you for the comment. The purpose of this callback is to initialize the matching so the accurate value is not necessary. I think just estimating height from x and y is enough because it is working well to start localization.

asimay, 2017-12-19:
ok。 @YamatoAndo @kitsukawa ，how about set z to 0?

YamatoAndo, 2017-12-22:
@asimay  If you do not use get_height option, the z value will be 0.
(Although it is not necessarily 0 because TF conversion is included at line 584)

closed 2018-01-30 by amc-nu