autowarefoundation/autoware #989: bug in Ndt_matching

asimay, 2017-12-19:
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp#L1071

# Bug
## Expected Behavior
at line 1071,  it seems no matter `use_predict_pose`  is, it always will be set to ` 0`.
so from line 1073 to 1091, it always run `if (use_predict_pose == 0)`  block.

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

asimay, 2017-12-19:
if you can send me more detail info of the points_callback() function logic, I could review more detail.

YamatoAndo, 2017-12-19:
@asimay  I thought that was strange, too.

@kitsukawa What do you think?
If it is always 0 and there is no problem, I'll delete the useless code and fix it

kitsukawa, 2017-12-19:
@YamatoAndo @asimay use_predict_pose is set 0 on purpose to disable temporarily. I tried to use predict pose to keep localization when matching fails, but it didn't work. So can you comment out the code related to predict_pose?

asimay, 2017-12-20:
how about use probability to predict pose? Now AI is so popular in the world.